SiLab-Bonn/pyBAR #6: RX sync lost

DavidLP, 2015-03-20:
Long runs producing lots of data (>> 10 threshold scans) have a high propability to abort due to lost RX sync. The LEDs of the MIO freeze; maybe indicating firmware bug?

LOG:
2015-03-19 17:57:10,148 - root - [INFO    ] (MainThread) Scan step 42 (PlsrDAC 198)
2015-03-19 17:57:10,151 - root - [INFO    ] (MainThread) Changing scan parameter(s): PlsrDAC=198
2015-03-19 17:57:10,153 - root - [INFO    ] (MainThread) Starting FIFO readout...
2015-03-19 17:57:10,153 - root - [INFO    ] (MainThread) Resetting SRAM FIFO: size = 0
2015-03-19 17:57:10,401 - root - [INFO    ] (MainThread) 100 injection(s): mask step 0 [0 - 2]
2015-03-19 17:57:11,042 - root - [INFO    ] (MainThread) 100 injection(s): mask step 1 [0 - 2]
2015-03-19 17:57:11,404 - root - [ERROR   ] (WatchdogThread) No RX sync. Aborting run...
2015-03-19 17:57:11,446 - root - [INFO    ] (MainThread) Sending global configuration to FE
2015-03-19 17:57:21,460 - root - [INFO    ] (MainThread) Stopped FIFO readout

laborleben, 2015-03-20:
fixed in Basil, new firmware was pushed

closed 2015-03-20 by laborleben

DavidLP, 2015-03-20:
RX errors e.g. due to ESD cannot be avoided. Longer runs have a higher propability that RX errors occur. These errors can be handled by putting 'reset_rx_on_error': True in the run_config. See for example [calibrate_threshold.py](https://github.com/SiLab-Bonn/pyBAR/blob/development/pybar/scans/calibrate_threshold.py)