SiLab-Bonn/pyBAR #7: S Curve fit failed

lauramari, 2015-03-30:
Fast Threshold Scan  aborted;  not enough points to fit S_Curve

Solved by updating; numpy 1.9.0 to numpy 1.9.2 ;Matplotlib 1.4.0 to Matplotlib 1.4.3
and Scipy 0.14.0 to 0.15.1

laborleben, 2015-03-30:
I'm afraid that more people will run into this. "conda update --all" is needed, I guess.

DavidLP, 2015-03-30:
Since the solution is already given I close the issue. Is there a way to make this post 'sticky', since the info is important?

closed 2015-03-30 by DavidLP

laborleben, 2015-03-30:
good point. Need to be in the documentation since anaconda is not using the latest packages by default.

DavidLP, 2015-03-30:
Ok I add it to wiki

DavidLP, 2015-03-30:
Well, actually Ill not add it to the wiki. It is anaconda related. We use miniconda now, that should use the latest packages, I guess. Needs cross check though...