SiLab-Bonn/pyBAR #8: USBError raised in Calibrate hit_or

lauramari, 2015-03-30:
hit_or Calibration raises USBError;

Traceback (most recent call last):
  File "C:\Users\student\git\pyBAR\pybar\run_manager.py", line 111, in run
    self.do_run()
  File "C:\Users\student\git\pyBAR\pybar\fei4_run_base.py", line 248, in do_run
    self.scan()
  File "C:\Users\student\git\pyBAR\pybar\scans\calibrate_hit_or.py", line 99, in scan
    self.dut['tdc_rx2']['ENABLE'] = False
  File "c:\users\student\git\basil\host\basil\HL\RegisterHardwareLayer.py", line 244, in __setitem__
    return self._set(name, value)
  File "c:\users\student\git\basil\host\basil\HL\RegisterHardwareLayer.py", line 234, in _set
    self.set_value(value, **descr)
  File "c:\users\student\git\basil\host\basil\HL\RegisterHardwareLayer.py", line 94, in set_value
    self._intf.write(self._base_addr + addr, data=array.array('B', reg.tobytes()))
  File "c:\users\student\git\basil\host\basil\TL\SiUsb.py", line 69, in write
    self._sidev.WriteExternal(addr - self.BASE_ADDRESS_EXTERNAL, data)
  File "C:\Users\student\Anaconda\lib\site-packages\SiLibUSB\siusbdevice.py", line 201, in WriteExternal
    self._write(self.SUR_TYPE_EXTERNAL, address, data)
  File "C:\Users\student\Anaconda\lib\site-packages\SiLibUSB\siusbdevice.py", line 241, in _write
    self._write_single(stype, addr, data[val:val + max_size])
  File "C:\Users\student\Anaconda\lib\site-packages\SiLibUSB\siusbdevice.py", line 251, in _write_single
    self.dev.write(ep, data[val:val + max_size])
  File "C:\Users\student\Anaconda\lib\site-packages\usb\core.py", line 885, in write
    self.__get_timeout(timeout)
  File "C:\Users\student\Anaconda\lib\site-packages\usb\backend\libusb1.py", line 808, in bulk_write
    timeout)
  File "C:\Users\student\Anaconda\lib\site-packages\usb\backend\libusb1.py", line 904, in __write
    _check(retval)
  File "C:\Users\student\Anaconda\lib\site-packages\usb\backend\libusb1.py", line 588, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
USBError: [Errno 10060] Operation timed out

laborleben, 2015-03-30:
it is hard to tell what is the reason for these backend error. Any specific observation?

DavidLP, 2015-03-30:
I cannot trigger this error under Linux, where I run this exact scan several times in the last days. Might though be libusb windows related? Maybe the time out is too strict? Its also a quite slow PC where this happended.

laborleben, 2015-03-30:
ah right, this remembers me to make the timeout less strict. This will certainly fix the issue. Laura can you check? The timeout can be set in pySiLibUsb self.intf.write() and self.intf.read() functions.

lauramari, 2015-03-30:
I managed by setting the _DEFAULT_TIMEOUT in core.py to 1000000. (It was
1000 previously) & USBError is not raised anymore.

But I wasn't able to set timeout in write() and read() functions...

DavidLP, 2015-03-30:
Ok good to know. But you changed a python package ;-). We should do this within pySiLibUsb I guess.

lauramari, 2015-03-31:
Actually, the problem is not solved, only the calibration rans longer
before the USBError is raised.
I was not sure how to set in pySiLibUsb. Is it in these functions in the
libusb1.py ?

    def __write(self, fn, dev_handle, ep, intf, data, timeout):
        address, length = data.buffer_info()
        length *= data.itemsize
        transferred = c_int()
        retval = fn(dev_handle.handle,
                  ep,
                  cast(address, POINTER(c_ubyte)),
                  length,
                  byref(transferred),
                  timeout)
        # do not assume LIBUSB_ERROR_TIMEOUT means no I/O.
        if not (transferred.value and retval == LIBUSB_ERROR_TIMEOUT):
            _check(retval)

        return transferred.value

    def __read(self, fn, dev_handle, ep, intf, buff, timeout):
        address, length = buff.buffer_info()
        length *= buff.itemsize
        transferred = c_int()
        retval = fn(dev_handle.handle,
                  ep,
                  cast(address, POINTER(c_ubyte)),
                  length,
                  byref(transferred),
                  timeout)
        # do not assume LIBUSB_ERROR_TIMEOUT means no I/O.
        if not (transferred.value and retval == LIBUSB_ERROR_TIMEOUT):
            _check(retval)
        return transferred.value

laborleben, 2015-03-31:
Can it be some broken USB driver?

Have you tried to install the latest chipset driver for your motherboard and USB driver if available?

DavidLP, 2015-04-12:
I have the same error using windows 7. In very long scans operation timeout is likely:

2015-04-12 09:18:26,842 - root - [ERROR   ] (MainThread) Unexpected exception during run 1: Tracebac
  File "c:\users\davidlp\git\pybar\pybar\run_manager.py", line 111, in run
    self.do_run()
  File "c:\users\davidlp\git\pybar\pybar\fei4_run_base.py", line 265, in do_run
    self.scan()
  File "c:\users\davidlp\git\pybar\pybar\scans\scan_ext_trigger_gdac.py", line 47, in scan
    ExtTriggerScan.scan(self)
  File "c:\users\davidlp\git\pybar\pybar\scans\scan_ext_trigger.py", line 87, in scan
    self.stop(msg='Trigger limit was reached: %i' % self.max_triggers)
  File "C:\Anaconda\lib\contextlib.py", line 24, in __exit__
    self.gen.next()
  File "c:\users\davidlp\git\pybar\pybar\fei4_run_base.py", line 364, in readout
    self.stop_readout()
  File "c:\users\davidlp\git\pybar\pybar\scans\scan_ext_trigger.py", line 130, in stop_readout
    self.dut['tdc_rx2']['ENABLE'] = False
  File "c:\users\davidlp\git\basil\host\basil\HL\RegisterHardwareLayer.py", line 244, in __setitem__
    return self._set(name, value)
  File "c:\users\davidlp\git\basil\host\basil\HL\RegisterHardwareLayer.py", line 234, in _set
    self.set_value(value, **descr)
  File "c:\users\davidlp\git\basil\host\basil\HL\RegisterHardwareLayer.py", line 90, in set_value
    ret = self._intf.read(self._base_addr + addr, size=div)
  File "c:\users\davidlp\git\basil\host\basil\TL\SiUsb.py", line 77, in read
    data = self._sidev.ReadExternal(addr - self.BASE_ADDRESS_EXTERNAL, size)
  File "C:\Anaconda\lib\site-packages\SiLibUSB\siusbdevice.py", line 204, in ReadExternal
    return self._read(self.SUR_TYPE_EXTERNAL, address, size)
  File "C:\Anaconda\lib\site-packages\SiLibUSB\siusbdevice.py", line 262, in _read
    ret += self._read_single(stype, val, size - val + addr)
  File "C:\Anaconda\lib\site-packages\SiLibUSB\siusbdevice.py", line 269, in _read_single
    self._write_sur(stype, self.SUR_DIR_IN, addr, size)
  File "C:\Anaconda\lib\site-packages\SiLibUSB\siusbdevice.py", line 285, in _write_sur
    self.dev.write(self.SUR_CONTROL_PIPE, ar)
  File "C:\Anaconda\lib\site-packages\usb\core.py", line 878, in write
    self.__get_timeout(timeout)
  File "C:\Anaconda\lib\site-packages\usb\backend\libusb1.py", line 768, in bulk_write
    timeout)
  File "C:\Anaconda\lib\site-packages\usb\backend\libusb1.py", line 864, in __write
    _check(retval)
  File "C:\Anaconda\lib\site-packages\usb\backend\libusb1.py", line 560, in _check
    raise USBError(_str_error[ret], ret, _libusb_errno[ret])
USBError: [Errno 10060] Operation timed out

laborleben, 2015-05-09:
I will close this since I received no feedback.

DavidLP, 2015-05-10:
It is still an open issue.

DavidLP, 2015-05-10:
2015-05-10 13:34:28,073 - root - [INFO    ] (MainThread) Taking data...
2015-05-10 13:46:36,717 - root - [ERROR   ] (ReadoutThread) [Errno 32] Pipe error. Aborting run...
2015-05-10 13:46:36,717 - root - [WARNING ] (MainThread) FIFO stop timeout
2015-05-10 13:46:36,717 - root - [ERROR   ] (WatchdogThread) [Errno 5] Input/Output Error
2015-05-10 13:46:36,742 - root - [ERROR   ] (WatchdogThread) [Errno 5] Input/Output Error
2015-05-10 13:46:36,742 - root - [ERROR   ] (WatchdogThread) [Errno 5] Input/Output Error
2015-05-10 13:46:36,746 - root - [ERROR   ] (WatchdogThread) [Errno 5] Input/Output Error
2015-05-10 13:46:36,746 - root - [INFO    ] (WatchdogThread) Data queue size: 0
Exception in thread WatchdogThread:
Traceback (most recent call last):
  File "C:\Miniconda\lib\threading.py", line 810, in __bootstrap_inner
    self.run()
  File "C:\Miniconda\lib\threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "c:\users\davidlp\git\pybar\pybar\daq\fifo_readout.py", line 238, in watchdog
    self.errback(sys.exc_info())
  File "c:\users\davidlp\git\pybar\pybar\fei4_run_base.py", line 334, in handle_err
    self.fifo_readout.print_readout_status()
  File "c:\users\davidlp\git\pybar\pybar\daq\fifo_readout.py", line 154, in print_readout_status
    logging.info('SRAM FIFO size: %d', self.dut['SRAM']['FIFO_SIZE'])
  File "c:\users\davidlp\git\basil\host\basil\HL\RegisterHardwareLayer.py", line 241, in __getitem__
    return self._get(name)
  File "c:\users\davidlp\git\basil\host\basil\HL\RegisterHardwareLayer.py", line 214, in _get
    ret_val = self.get_value(**descr)
  File "c:\users\davidlp\git\basil\host\basil\HL\RegisterHardwareLayer.py", line 116, in get_value
    ret = self._intf.read(self._base_addr + addr, size=div)
  File "c:\users\davidlp\git\basil\host\basil\TL\SiUsb.py", line 77, in read
    data = self._sidev.ReadExternal(addr - self.BASE_ADDRESS_EXTERNAL, size)
  File "C:\Miniconda\lib\site-packages\SiLibUSB\siusbdevice.py", line 211, in ReadExternal
    return self._read(self.SUR_TYPE_EXTERNAL, address, size)
  File "C:\Miniconda\lib\site-packages\SiLibUSB\siusbdevice.py", line 269, in _read
    ret += self._read_single(stype, val, size - val + addr)
  File "C:\Miniconda\lib\site-packages\SiLibUSB\siusbdevice.py", line 276, in _read_single
    self._write_sur(stype, self.SUR_DIR_IN, addr, size)
  File "C:\Miniconda\lib\site-packages\SiLibUSB\siusbdevice.py", line 292, in _write_sur
    self.dev.write(self.SUR_CONTROL_PIPE, ar)
  File "C:\Miniconda\lib\site-packages\usb\core.py", line 898, in write
    self.__get_timeout(timeout)
  File "C:\Miniconda\lib\site-packages\usb\backend\libusb1.py", line 816, in bulk_write
    timeout)
  File "C:\Miniconda\lib\site-packages\usb\backend\libusb1.py", line 912, in __write
    _check(retval)
  File "C:\Miniconda\lib\site-packages\usb\backend\libusb1.py", line 592, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
USBError: [Errno 5] Input/Output Error

themperek, 2015-05-10:
Does one need to power cycle MIO card after this?

DavidLP, 2015-05-10:
Yes

themperek, 2015-05-10:
Then most likely this is FX2 firmware. Some combination of commands make it fail. 
Does this happens on Linux/Mac to?

DavidLP, 2015-05-10:
Have not seen it there. yet.

laborleben, 2015-06-10:
the error was due to a timeout in dev.read(). This was fixed in pySiLibUSB 2.0.4.

Installation guide: https://silab-redmine.physik.uni-bonn.de/projects/pysilibusb/wiki

DavidLP, 2015-06-26:
No, it is not fixed and still an open issue.

laborleben, 2015-06-29:
did increasing the dev.write timeout improved the situation?

DavidLP, 2015-06-30:
Hard to tell yet. It's a rare phenomenon, I need more statistics to judge.

laborleben, 2015-07-08:
increased timeout, added timeout for dev.write().

Please update to pySiLibUSB 2.0.5.

closed 2015-07-08 by laborleben