SiLab-Bonn/pyBAR #11: Malloc using analysis extension

DavidLP, 2015-04-26:
A potential runtime bug was reported on the Mailing list. The usage of the analysis library causes a malloc error on:

Ubuntu 14.04
Python 2.7.9 |Continuum Analytics, Inc.| (default, Mar  9 2015, 16:21:41) 
[GCC 4.4.7 20120313 (Red Hat 4.4.7-1)] on linux2

Example Output 1:
2015-04-24 18:19:18,658 - root - [INFO    ] (MainThread) Scan parameter(s) from raw data file(s): None
2015-04-24 18:19:18,659 - root - [INFO    ] (MainThread) Opening output PDF file: /home/sgrinstein/workspace/CNM103/module_test/7_module_test_digital_scan.pdf
2015-04-24 18:19:18,661 - root - [INFO    ] (MainThread) Interpreting raw data file(s): /home/sgrinstein/workspace/CNM103/module_test/7_module_test_digital_scan.h5
python: malloc.c:2372: sysmalloc: Assertion `(old_top == (((mbinptr) (((char *) &((av)->bins[((1) - 1) * 2])) - __builtin_offsetof (struct malloc_chunk, fd)))) && old_size == 0) || ((unsigned long) (old_size) >= (unsigned long)((((__builtin_offsetof (struct malloc_chunk, fd_nextsize))+((2 *(sizeof(size_t))) - 1)) & ~((2 *(sizeof(size_t))) - 1))) && ((old_top)->size & 0x1) && ((unsigned long) old_end & pagemask) == 0)' failed.

Example Output 2:
2015-04-24 18:42:04,756 - root - [INFO    ] (MainThread) Create selected cluster histograms
2015-04-24 18:42:04,761 - root - [INFO    ] (MainThread) Closing output PDF file: test_analysis//unit_test_data_1.pdf
python: malloc.c:3695: _int_malloc: Assertion `(unsigned long) (size) >= (unsigned long) (nb)' failed.

Neither the build and testing on Windows 7 / Windows Server 32/64 bit, Ubuntu 12.04 LTS Server 64 bit and Ubuntu 14.04 LTS 64 bit showed this error before.

A Valgrind report was created showing one memory leak in Clusterizer::allocateResultHistograms().

DavidLP, 2015-04-26:
Valgrind Output for test_analysis unittest:

==18195== Memcheck, a memory error detector
==18195== Copyright (C) 2002-2012, and GNU GPL'd, by Julian Seward et al.
==18195== Using Valgrind-3.8.1 and LibVEX; rerun with -h for copyright info
==18195== Command: python test_analysis.py
==18195== 
==18195== Invalid write of size 8
==18195==    at 0x2AD11EAD: Clusterizer::clearResultHistograms() (Clusterizer.cpp:717)
==18195==    by 0x2AD12333: Clusterizer::reset() (Clusterizer.cpp:183)
==18195==    by 0x2AD13FF9: Clusterizer::Clusterizer() (Clusterizer.cpp:14)
==18195==    by 0x2AD05B7C: __pyx_tp_new_5pybar_8analysis_16RawDataConverter_16data_clusterizer_PyDataClusterizer(_typeobject*, _object*, _object*) (data_clusterizer.cpp:1978)
==18195==    by 0x4CB91C2: type_call (typeobject.c:729)
==18195==    by 0x4C56322: PyObject_Call (abstract.c:2529)
==18195==    by 0x4D05361: PyEval_EvalFrameEx (ceval.c:4251)
==18195==    by 0x4D065FF: PyEval_EvalFrameEx (ceval.c:4119)
==18195==    by 0x4D065FF: PyEval_EvalFrameEx (ceval.c:4119)
==18195==    by 0x4D07C6D: PyEval_EvalCodeEx (ceval.c:3265)
==18195==    by 0x4C85957: function_call (funcobject.c:526)
==18195==    by 0x4C56322: PyObject_Call (abstract.c:2529)
==18195==    by 0x4D04864: PyEval_EvalFrameEx (ceval.c:4346)
==18195==    by 0x4D07C6D: PyEval_EvalCodeEx (ceval.c:3265)
==18195==    by 0x4C85850: function_call (funcobject.c:526)
==18195==    by 0x4C56322: PyObject_Call (abstract.c:2529)
==18195==    by 0x4C6895E: instancemethod_call (classobject.c:2602)
==18195==    by 0x4C56322: PyObject_Call (abstract.c:2529)
==18195==    by 0x4CC2AAB: slot_tp_call (typeobject.c:5432)
==18195==    by 0x4C56322: PyObject_Call (abstract.c:2529)
==18195==    by 0x4D05361: PyEval_EvalFrameEx (ceval.c:4251)
==18195==    by 0x4D065FF: PyEval_EvalFrameEx (ceval.c:4119)
==18195==    by 0x4D07C6D: PyEval_EvalCodeEx (ceval.c:3265)
==18195==    by 0x4D07D81: PyEval_EvalCode (ceval.c:667)
==18195==    by 0x4D27B9F: PyRun_FileExFlags (pythonrun.c:1371)
==18195==    by 0x4D27D7E: PyRun_SimpleFileExFlags (pythonrun.c:949)
==18195==    by 0x4D3D663: Py_Main (main.c:645)
==18195==    by 0x3D4701D9F3: (below main) (in /lib64/libc-2.5.so)
==18195==  Address 0x21f50148 is 524,280 bytes inside a block of size 524,284 alloc'd
==18195==    at 0x4A08392: operator new[](unsigned long) (in /opt/rh/devtoolset-2/root/usr/lib64/valgrind/vgpreload_memcheck-amd64-linux.so)
==18195==    by 0x2AD13C0C: Clusterizer::allocateResultHistograms() (Clusterizer.cpp:703)
==18195==    by 0x2AD13FE9: Clusterizer::Clusterizer() (Clusterizer.cpp:12)
==18195==    by 0x2AD05B7C: __pyx_tp_new_5pybar_8analysis_16RawDataConverter_16data_clusterizer_PyDataClusterizer(_typeobject*, _object*, _object*) (data_clusterizer.cpp:1978)
==18195==    by 0x4CB91C2: type_call (typeobject.c:729)
==18195==    by 0x4C56322: PyObject_Call (abstract.c:2529)
==18195==    by 0x4D05361: PyEval_EvalFrameEx (ceval.c:4251)
==18195==    by 0x4D065FF: PyEval_EvalFrameEx (ceval.c:4119)
==18195==    by 0x4D065FF: PyEval_EvalFrameEx (ceval.c:4119)
==18195==    by 0x4D07C6D: PyEval_EvalCodeEx (ceval.c:3265)
==18195==    by 0x4C85957: function_call (funcobject.c:526)
==18195==    by 0x4C56322: PyObject_Call (abstract.c:2529)
==18195==    by 0x4D04864: PyEval_EvalFrameEx (ceval.c:4346)
==18195==    by 0x4D07C6D: PyEval_EvalCodeEx (ceval.c:3265)
==18195==    by 0x4C85850: function_call (funcobject.c:526)
==18195==    by 0x4C56322: PyObject_Call (abstract.c:2529)
==18195==    by 0x4C6895E: instancemethod_call (classobject.c:2602)
==18195==    by 0x4C56322: PyObject_Call (abstract.c:2529)
==18195==    by 0x4CC2AAB: slot_tp_call (typeobject.c:5432)
==18195==    by 0x4C56322: PyObject_Call (abstract.c:2529)
==18195==    by 0x4D05361: PyEval_EvalFrameEx (ceval.c:4251)
==18195==    by 0x4D065FF: PyEval_EvalFrameEx (ceval.c:4119)
==18195==    by 0x4D07C6D: PyEval_EvalCodeEx (ceval.c:3265)
==18195==    by 0x4D07D81: PyEval_EvalCode (ceval.c:667)
==18195==    by 0x4D27B9F: PyRun_FileExFlags (pythonrun.c:1371)
==18195==    by 0x4D27D7E: PyRun_SimpleFileExFlags (pythonrun.c:949)
==18195==    by 0x4D3D663: Py_Main (main.c:645)
==18195==    by 0x3D4701D9F3: (below main) (in /lib64/libc-2.5.so)
==18195== 
2015-04-25 20:48:40,370 - root - [INFO    ] (MainThread) Scan parameter(s) from raw data file(s): None
2015-04-25 20:48:40,434 - root - [INFO    ] (MainThread) Opening output PDF file: test_analysis//unit_test_data_1.pdf
2015-04-25 20:48:40,994 - root - [INFO    ] (MainThread) Interpreting raw data file(s): test_analysis//unit_test_data_1.h5
2015-04-25 20:48:41,182 - root - [INFO    ] (MainThread) Interpreting...
[... 436 lines cut ...]
	27	0
	28	0
	29	0
	30	0
	31	0
==18195== 
==18195== HEAP SUMMARY:
==18195==     in use at exit: 19,967,035 bytes in 17,974 blocks
==18195==   total heap usage: 6,298,724 allocs, 6,280,750 frees, 33,251,246,614 bytes allocated
==18195== 

...

==18195== 
==18195== LEAK SUMMARY:
==18195==    definitely lost: 192 bytes in 6 blocks
==18195==    indirectly lost: 0 bytes in 0 blocks
==18195==      possibly lost: 480,647 bytes in 714 blocks
==18195==    still reachable: 19,486,164 bytes in 17,253 blocks
==18195==         suppressed: 32 bytes in 1 blocks
==18195== Reachable blocks (those to which a pointer was found) are not shown.
==18195== To see them, rerun with: --leak-check=full --show-reachable=yes
==18195== 
==18195== For counts of detected and suppressed errors, rerun with: -v
==18195== ERROR SUMMARY: 434 errors from 370 contexts (suppressed: 24303 from 358)

themperek, 2015-04-26:
Just small remark. This is not memory leak.

"the program wrote to some memory it should not have due to a heap block overrun."
see http://valgrind.org/docs/manual/quick-start.html

DavidLP, 2015-04-26:
Yes, it is not a memory leak but a wrong memory access. I hope I fixed it in the last comit. Mabe you can check?

themperek, 2015-04-26:
Seems better: https://dl.dropboxusercontent.com/u/2937630/Valgrind.pybar.log3.txt

Was original problem on 32 or 64 bit?

DavidLP, 2015-04-26:
thank you! Ill ask if it was 32 bit or not.

themperek, 2015-04-27:
Cannot reproduce on ubuntu 14.04.2 32bit but one test fails:
== on floats?

```
FAIL: test_threshold_analysis (__main__.TestAnalysis)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "test_analysis.py", line 255, in test_threshold_analysis
    self.assertTrue(data_equal, msg=error_msg)
AssertionError: HistNoise: Sum first array: 89077.2468745, Sum second array: 89077.2468745
HistThreshold: Sum first array: 719849.92, Sum second array: 719849.92

----------------------------------------------------------------------
Ran 16 tests in 137.341s

FAILED (failures=1)
```

Full log: https://dl.dropboxusercontent.com/u/2937630/pybar.ubuntu.14.04.x86.txt

DavidLP, 2015-04-28:
Noise is a fit result. I guess some sub sub digits can be slightly different. To add a np.allclose instead of checking for exact similarity for fitted results makes more sense.

DavidLP, 2015-04-28:
The malloc error does not happen anymore after last [changes ](https://github.com/SiLab-Bonn/pyBAR/commit/51b19bc52c10bf24692cbb593eebc9841920dc4f)  thus the bug is fixed.

closed 2015-04-28 by DavidLP