SiLab-Bonn/pyBAR #16: External trigger scan crash

lauramari, 2015-05-19:
The scan_external_trigger scan crashes with the message attached below.

Traceback (most recent call last):
  File "C:\Users\student\git\pyBAR\pybar\run_manager.py", line 111, in run
    self.do_run()
  File "C:\Users\student\git\pyBAR\pybar\fei4_run_base.py", line 295, in do_run
    self.scan()
  File "C:\Users\student\git\pyBAR\pybar\scans\scan_ext_trigger.py", line 74, in scan
    with self.readout(**self.scan_parameters._asdict()):
  File "C:\Users\student\Anaconda\lib\contextlib.py", line 17, in __enter__
    return self.gen.next()
  File "C:\Users\student\git\pyBAR\pybar\fei4_run_base.py", line 392, in readout
    self.start_readout(*args, **kwargs)
  File "C:\Users\student\git\pyBAR\pybar\scans\scan_ext_trigger.py", line 120, in start_readout
    self.dut['TLU']['EN_INVERT_TRIGGER'] = 0 if self.trigger_pos_edge else 1
  File "c:\users\student\git\basil\host\basil\HL\RegisterHardwareLayer.py", line 244, in __setitem__
    return self._set(name, value)
  File "c:\users\student\git\basil\host\basil\HL\RegisterHardwareLayer.py", line 222, in _set
    descr = deepcopy(self._registers[reg]['descr'])
KeyError: 'EN_INVERT_TRIGGER'

laborleben, 2015-05-19:
please update pyBAR *and* basil to the latest development version

lauramari, 2015-05-19:
It was already updated.
And I notice the scan only crashes when I enable the tdc to True.

laborleben, 2015-05-20:
this is fixed now.

Please use dut_configuration_mio.yaml to set up trigger (mode). Comments on the settings are available inside the file.

closed 2015-05-20 by laborleben