SiLab-Bonn/pyBAR #19: Scans hang with no reported error/without crashing

lauramari, 2015-06-01:
(All) Scans hang after logging info 'Success!' on the Console after programming of the FPGA.

I have tried replacing the production module adapter but the issue is not resolved. Below is the output from the console

2015-06-01 18:53:13,996 - root - [INFO    ] (MainThread) Initializing HitOrCalibration
2015-06-01 18:53:13,996 - root - [INFO    ] (MainThread) Press Ctrl-C to stop run
2015-06-01 18:53:14,006 - root - [INFO    ] (MainThread) Starting run #309 (HitOrCalibration) in ..\module_2156
2015-06-01 18:53:14,016 - root - [INFO    ] (MainThread) Scan parameter(s): column=None, row=None, PlsrDAC=[40, 50, 60, 80, 130, 180, 230, 280, 340, 440, 540, 640, 740]
2015-06-01 18:53:14,157 - root - [INFO    ] (MainThread) Found USB board(s): USBpix with ID 213 (FW 15)
2015-06-01 18:53:14,157 - root - [INFO    ] (MainThread) Programming FPGA: mio.bit...
2015-06-01 18:53:17,417 - root - [INFO    ] (MainThread) Success!

laborleben, 2015-06-01:
is it then crashing without any further information? Or is it just hanging in a loop?

lauramari, 2015-06-01:
It just hangs in a loop. There's no crash information given.

closed 2015-06-02 by lauramari