SiLab-Bonn/pyBAR #23: Init and close of devices

DavidLP, 2015-06-29:
There is a potential bug by design in how devices are treated in the [fei4_run_base](https://github.com/SiLab-Bonn/pyBAR/blob/development/pybar/fei4_run_base.py). 
While devices are _initialized only once_ in the pre_run method when [isinstance(self.conf['dut'], Dut)](https://github.com/SiLab-Bonn/pyBAR/blob/development/pybar/fei4_run_base.py#L250) is true they are _always closed_ in the [cleanup_run](https://github.com/SiLab-Bonn/pyBAR/blob/development/pybar/fei4_run_base.py#L348) method. 
When writing a prim list this means that a Serial port is for example closed and not reopened when rerunning the scan. 
How to deal with this?

laborleben, 2015-06-30:
will send you files.

DavidLP, 2015-06-30:
The change in fei4_run_base I also tried yesterday, with the same result, that pyBar hangs and is not responding anymore. The latest log lines are:

2015-06-30 10:54:41,936 - root - [INFO    ] (MainThread) Found adapter card: Single Chip Adapter Card with ID 74
2015-06-30 10:54:41,946 - root - [INFO    ] (MainThread) Initializing fei4_rx from module basil.HL.fei4_rx (Version: 2)

laborleben, 2015-06-30:
did you also use SiUsb.py?

DavidLP, 2015-06-30:
Yes I did.

laborleben, 2015-07-08:
2 possibilities:

1. only close USB device at the end of each run. All other devices are kept open. Potential hazard: Devices are not properly closed after the last run (but this may depend on the python implementation).

2. close all devices at the end of each run and initialize at the beginning of each run. Potential problem: This may slow down things a little bit.

DavidLP, 2015-07-09:
I'm fine with solution 1. since all implementations we use (and I know, e.g. pySerial, pyVisa, pyZeroMq) close the device when the object is destroyed. Also the OS should always automatically take the control of the device if the program is closes / crashes ?

commit 759f08ca (2015-07-13): ENH: properly close all SiUsb interfaces, while not calling close on other modules, fixes #23
closed 2015-07-13 by laborleben