SiLab-Bonn/pyBAR #24: Sending of data with parameters as JSON

DavidLP, 2015-06-30:
When parameters as an ordered dict are send an exception is thrown that this is not JSON serializable.

laborleben, 2015-07-08:
more details needed.

DavidLP, 2015-07-08:
No

DavidLP, 2015-07-08:
-------------------- Run 11 --------------------
Traceback (most recent call last):
  File "c:\users\silab\workspace_pydev\pybar\pybar\run_manager.py", line 112, in run
    self.do_run()
  File "C:\Miniconda\lib\contextlib.py", line 24, in __exit__
    self.gen.next()
  File "c:\users\silab\workspace_pydev\pybar\pybar\run_manager.py", line 160, in _run
    self.post_run()
  File "c:\users\silab\workspace_pydev\pybar\pybar\daq\fifo_readout.py", line 221, in worker
    self.callback(data)
  File "c:\users\silab\workspace_pydev\pybar\pybar\scans\calibrate_hit_or.py", line 191, in handle_data
    self.raw_data_file.append_item(data, scan_parameters=self.scan_parameters._asdict(), new_file=['column'], flush=False)  # Create new file for each scan parameter change
  File "c:\users\silab\workspace_pydev\pybar\pybar\daq\fei4_raw_data.py", line 200, in append_item
    send_data(self.socket, data_tuple, self.scan_parameters)
  File "c:\users\silab\workspace_pydev\pybar\pybar\daq\fei4_raw_data.py", line 40, in send_data
    socket.send_json(data_meta_data, flags=zmq.SNDMORE | zmq.NOBLOCK)
  File "C:\Miniconda\lib\site-packages\zmq\sugar\socket.py", line 451, in send_json
    msg = jsonapi.dumps(obj, **kwargs)
  File "C:\Miniconda\lib\site-packages\zmq\utils\jsonapi.py", line 40, in dumps
    s = jsonmod.dumps(o, **kwargs)
  File "C:\Miniconda\lib\json\__init__.py", line 250, in dumps
    sort_keys=sort_keys, **kw).encode(obj)
  File "C:\Miniconda\lib\json\encoder.py", line 207, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "C:\Miniconda\lib\json\encoder.py", line 270, in iterencode
    return _iterencode(o, 0)
  File "C:\Miniconda\lib\json\encoder.py", line 184, in default
    raise TypeError(repr(o) + " is not JSON serializable")
TypeError: 55 is not JSON serializable

-------------------- Run 12 --------------------
Traceback (most recent call last):
  File "c:\users\silab\workspace_pydev\pybar\pybar\run_manager.py", line 112, in run
    self.do_run()
  File "c:\users\silab\workspace_pydev\pybar\pybar\fei4_run_base.py", line 304, in do_run
    with open_raw_data_file(filename=self.output_filename, mode='w', title=self.run_id, register=self.register, conf=self.conf, run_conf=self.run_conf, scan_parameters=self.scan_parameters._asdict(), socket_addr=self.socket_addr) as self.raw_data_file:
  File "c:\users\silab\workspace_pydev\pybar\pybar\daq\fei4_raw_data.py", line 57, in open_raw_data_file
    return RawDataFile(filename=filename, mode=mode, title=title, register=register, conf=conf, run_conf=run_conf, scan_parameters=scan_parameters, socket_addr=socket_addr)
  File "c:\users\silab\workspace_pydev\pybar\pybar\daq\fei4_raw_data.py", line 100, in __init__
    self.socket.bind(socket_addr)
  File "zmq/backend/cython/socket.pyx", line 489, in zmq.backend.cython.socket.Socket.bind (zmq\backend\cython\socket.c:4824)
  File "zmq/backend/cython/checkrc.pxd", line 25, in zmq.backend.cython.checkrc._check_rc (zmq\backend\cython\socket.c:7055)
    raise ZMQError(errno)
ZMQError: Address in use

laborleben, 2015-07-13:
this is due to a a bug in calibrate_hit_or.py. np.int64 was used for scan parameter instead int.

closed 2015-07-13 by laborleben