SiLab-Bonn/pyBAR #26: Long LVDS path "No RX sync"

trogonpete, 2015-08-01:
(Using dev version of pyBAR at UH with USBpix and SEABAS2)

We have a 37m cable length between an FE-I4B chip board and USBpix/SEABAS2 readout board. We're testing with three different cat5e cables and one cat7. One or two of the four cables communicate successfully (cat5e), the others (including the cat7) fail to find RX sync:

"Run 131 was aborted: RX 8b10b error(s) detected"

or

"Run 129 was aborted: No RX sync"

We've probed the LVDS signals in all cables and there is no visible degradation over the 37m. Presumably the configs need some tuning for long cables but it seems odd that duplicates of the same cables give different results.

themperek, 2015-08-01:
I do not think anything can be done on FPGA side to improve this. At the moment one of 4 sampling points are selected and if signal is low quality can fail. One could do it more precise sampling point selection but is bit complicated. 

You could try to change LVDS driver parameters on FEI4 side: 
LVDSDrvIref, 
LvdsDrvSet06
LvdsDrvSet12
LvdsDrvSet30
LvdsDrvVos

You can also change receiving clock phase in small steps: https://github.com/SiLab-Bonn/pyBAR/blob/master/firmware/seabas2/src/clock_manager.v#L56
This may work but only for particular cable.

trogonpete, 2015-08-10:
Thanks for the help. We now can usually run with the long cables, but 8b10b errors are still common, causing unacceptable downtime. It seems like a total abort on a single 8b10b fault is unnecessary. Is better error handling realistic?

themperek, 2015-08-10:
You can try to comment checking this here:
https://github.com/SiLab-Bonn/pyBAR/blob/master/pybar/daq/fifo_readout.py#L233

One could make it a parameter long run @laborleben.

trogonpete, 2015-08-10:
Nice, that did the trick. We get occasional data corruption (a number of columns get knocked out midway through a run--always the same columns--and remain off throughout the run), but our downtime is much better. Thanks!

DavidLP, 2015-08-11:
For long runs there is a chance that a RX 8b10b error occurs. Therefore we implemented a feature that pyBAR handles this error and does not abort. 

In the run_config just set the parameter: [reset_rx_on_error](https://github.com/SiLab-Bonn/pyBAR/blob/development/pybar/scans/scan_ext_trigger.py#L33) to True

If this happens in short runs (e.g. threshold scans) you data transmission is bad and this should be solved on the hardware side.

trogonpete, 2015-08-27:
Thank you, that is very helpful. 

It's clear that we need a hardware upgrade for these long cables. For this, it would be very helpful to be able to run a simple communications diagnostic scan, just sending and receiving large blocks of data to/from the chip and counting errors. Do you have such a scan implemented already?

laborleben, 2015-09-01:
There is no such scan. A threshold scan and noise occupancy scan will generate large amount of data.

laborleben, 2015-09-01:
to enable this feature for all scans, just edit configuration.yaml and add them following lines:
```yaml
run_conf:
    reset_rx_on_error : False
```

DavidLP, 2015-09-01:
And it should be set to True ;-)

DavidLP, 2015-09-15:
The bad data transmission needs investigation of the SEABAS hardware developers and is not related to pyBAR. Thus I close this issue.

closed 2015-09-15 by DavidLP

themperek, 2016-10-13:
We found that the errors may come due to bad synchronization betwean CMD_DATA and CMD_CLK. By having more precise phase tuning we managed to operate very long (30m) cables. Some simple solution here: https://github.com/SiLab-Bonn/pyBAR/blob/beast/firmware/mmc3_beast_eth/src/mmc3_beast_eth.v#L425-L431

trogonpete, 2017-04-05:
Excellent, and thank you. This seems promising, but at the moment isn't working... we cannot achieve RX sync with a FE-I4 with the mmc3_beast_eth firmware. With mmc3_m26_eth we have no problem (but it doesn't have the clock phase fix), so it seems like this problem is unique to the BEAST firmware. Any ideas?

themperek, 2017-04-05:
The firmware is almost identical.
You can try to change the value of: CLK_DLY
here: https://github.com/SiLab-Bonn/pyBAR/blob/development/pybar/fei4_run_base.py#L201

DavidLP, 2017-04-05:
> we cannot achieve RX sync with a FE-I4 with the mmc3_beast_eth firmware.

While the firmware is quite identical the LVDS connection is very different between `mmc3_beast_eth` and `mmc3_m26_eth` as far as I remember, to match the BEAST stave. So with the same adapter I do not expect the `mmc3_beast_eth` firmware to work.

DavidLP, 2017-04-05:
If `rx_errors` only appear in scans where masks are shifted (=all tuning and calibration scans) but not in the data taking scans (e.g. `scan_ext_trigger_stop_mode`, `scan_fei4_self_trigger`, ...) then this is a hint for an insufficient powering of the FE. The FE has a bug that during mask shifting it has a state where the power consumption quickly doubles for a short time. If the power cables have a too high resistance this can lead to a short voltage drop putting the PLL out of phase creating `rx_errors`.

laborleben, 2017-04-05:
There is some confusion going on here, so please let me clarify:

1. The beast firmware is not pin compatible to the other firmwares. The beast firmware (MMC3_BEAST_ETH) uses a pin assignment (LVDS signals) which is solely compatible to the FANGS stave.
The other firmwares (MIO, MMC_M26_ETH, MMC3_QUAD_ETH) are supporting the standard pin assignment of the single chip/burn-in/quad-module adapter card (probably what you are using).
2. The MMC3_QUAD_ETH is probably what you want to use, which is currently only available in the multi_chip_sp branch of pyBAR. The firmware/brach is supporting the readout and configuration of multiple modules and is compatible the the standard pin assignment.
3. The clock to Rx data phase alignment is included in every firmware. Everything works well, when "Rx Sync OK" appears.
4. A bad Rx sync might have several reasons: Either the FE is not programmed correctly (actually a bad clock to Tx data phase alignment, which you might recognize from the power consumption and missing 8b10b idle pattern on the Rx line) or the Rx clock to data phase alignment is not working correctly (which is very unlikely, but in you case very likely, since you are using a long data cable).
5. The additional command delay, which is only implemented in the MMC3_BEAST_ETH firmware, is only necessary if you are programming multiple modules over a long cable, and in addition all modules share the same command data line (Tx, for configuring the modules) and input clock. An individual delay is added to the command output lines to satisfy each FE.
6. The command delay in the MMC3_BEAST_ETH firmware is currently commented out.

NOTE: The multi_chip_sp branch of pyBAR is currently under development. Please use MMC_M26_ETH when you are connecting a single module. Otherwise switch to the multi_chip_sp branch and use MMC3_QUAD_ETH firmware.

trogonpete, 2017-04-05:
OK, excellent, thanks for all the help. 

Our ultimate goal has been to independently trigger 8 FE-I4B chips on a single MMC3. Will mmc3_quad_eth eventually support this, or can it only independently trigger 4 chips?

themperek, 2017-04-05:
mmc3_quad_eth supports 8FEs at this point.

https://github.com/SiLab-Bonn/pyBAR/blob/multi_chip_sp/firmware/mmc3_quad_eth/src/mmc3_quad_eth.v

trogonpete, 2017-04-05:
Thanks. If I'm reading it correctly, the trigger is still taken from either RX0 or TLU. Is it possible to independently trigger?

laborleben, 2017-04-06:
OK, this is solved now.
@trogonpete : This is off-topic. Can you please open a new question/ticket/feature request about trigger?