SiLab-Bonn/pyBAR #29: max_trigger=None in eudaq_ex_trigger.py won't work?

thirono, 2015-09-12:
015-09-12 14:35:32,088 - root - [ERROR   ] (RunThread ) Unexpected exception during run 9: Traceback (most recent call last):
  File "/home/user/workspace/pyBAR_roi/branches/development/pybar/run_manager.py", line 121, in run
    self.do_run()
  File "/home/user/workspace/pyBAR_roi/branches/development/pybar/fei4_run_base.py", line 345, in do_run
    self.scan()
  File "eudaq_ext_trigger.py", line 54, in scan
    with self.readout(**self.scan_parameters._asdict()):
  File "/home/user/anaconda/lib/python2.7/contextlib.py", line 17, in __enter__
    return self.gen.next()
  File "/home/user/workspace/pyBAR_roi/branches/development/pybar/fei4_run_base.py", line 459, in readout
    self.start_readout(*args, **kwargs)
  File "/home/user/workspace/pyBAR_roi/branches/development/pybar/scans/scan_ext_trigger.py", line 117, in start_readout
    self.dut['TLU']['MAX_TRIGGERS'] = self.max_triggers
  File "/home/user/workspace/basil_itk/branches/development/host/basil/HL/RegisterHardwareLayer.py", line 244, in __setitem__
    return self._set(name, value)
  File "/home/user/workspace/basil_itk/branches/development/host/basil/HL/RegisterHardwareLayer.py", line 234, in _set
    self.set_value(value, **descr)
  File "/home/user/workspace/basil_itk/branches/development/host/basil/HL/RegisterHardwareLayer.py", line 93, in set_value
    reg[size + offset - 1:offset] = value
  File "/home/user/workspace/basil_itk/branches/development/host/basil/utils/BitLogic.py", line 97, in __setitem__
    bitarray.__setitem__(self, slc, item)
IndexError: bitarray or bool expected for slice assignment

laborleben, 2015-09-13:
do you use basil development (v2.1.2)? MAX_TRIGGERS is supported from v2.1.2.

laborleben, 2015-09-14:
Use MAX_TRIGGERS = 0 if you do not want to use that feature.

closed 2015-09-14 by laborleben