SiLab-Bonn/pyBAR #30: latest Miniconda gives errors

laborleben, 2015-10-12:
Traceback (most recent call last):
  File "/Users/jens/miniconda/lib/python2.7/site-packages/pyqtgraph/graphicsItems/ImageItem.py", line 309, in paint
    self.render()
  File "/Users/jens/miniconda/lib/python2.7/site-packages/pyqtgraph/graphicsItems/ImageItem.py", line 301, in render
    argb, alpha = fn.makeARGB(image.transpose((1, 0, 2)[:image.ndim]), lut=lut, levels=self.levels)
  File "/Users/jens/miniconda/lib/python2.7/site-packages/pyqtgraph/functions.py", line 942, in makeARGB
    data = rescaleData(data, scale/(maxVal-minVal), minVal, dtype=int)
  File "/Users/jens/miniconda/lib/python2.7/site-packages/pyqtgraph/functions.py", line 828, in rescaleData
    d2 *= scale
TypeError: Cannot cast ufunc multiply output from dtype('float64') to dtype('uint32') with casting rule 'same_kind'

laborleben, 2015-10-13:
I found the following solution: create conda environment with numpy=1.9
```
conda create -n pybar numpy=1.9 bitarray cython ipython matplotlib mock nose pandas pyserial pytables pyyaml pyzmq scipy sphinx
```
```
source activate pybar
```
install and (re-)compile all the other packages (including basil and pybar).

DavidLP, 2015-10-13:
So pyqtgraph is not compatible with newer numpy versions?

laborleben, 2015-10-13:
on a first view, yes

laborleben, 2015-10-13:
we need to change the manual, if this cant be solved.

DavidLP, 2015-10-13:
Wrote the developers a bug report:

https://github.com/pyqtgraph/pyqtgraph/issues/236

Since the bugs pile up in pyqtgraph we could think to write our own patch. Should not be difficult.

DavidLP, 2015-10-14:
If they do not fix it we can host a patched fork on our silab conda storage.

laborleben, 2015-10-14:
fork and make a pull request. Should be the easiest for the devs.

DavidLP, 2015-10-14:
They have 55 open pull requests. I think that will not work for us.

DavidLP, 2015-11-26:
PyQTGraph was forked and patched. The patch was delivered to PyQTgraph developers but they failed to [pull](https://github.com/pyqtgraph/pyqtgraph/pull/239) it. It might still take time until this is implemented, so we changed the [installation guide](https://github.com/SiLab-Bonn/pyBAR/wiki/Step-by-step-Installation-Guide) to use our patched fork. This is now our temporary solution.

closed 2015-11-26 by DavidLP