SiLab-Bonn/pyBAR #36: fix unit test in development

laborleben, 2016-02-18:
test_analysis.py is failing

DavidLP, 2016-02-18:
Seems to be conda related. With my python under windows it works.

DavidLP, 2016-02-18:
Travis does not fail

DavidLP, 2016-02-18:
It fails because of your bug comitted here: https://github.com/SiLab-Bonn/pyBAR/commit/946014cc8fff1693754f1f10bed4f92d48270871

What was the reason to do that?

DavidLP, 2016-02-18:
I revert.

laborleben, 2016-02-18:
The interface of the clusterizer changed. Probably I was to enthusiastic and deleted more than necessary.

DavidLP, 2016-02-18:
The API did not really change, but more needed features were implemented. The needed clusterizer version >= 2.3.0 is a released version hosted on PyPI. 
All the problems where there, because not the latest clusterizer version was used. The correct and working version was always mentioned in the requirenment.txt of pybar. I close this issue.

closed 2016-02-18 by DavidLP

laborleben, 2016-02-18:
there was version 2.2 still in my site-packages... funny that other packages were working ok.