SiLab-Bonn/pyBAR #41: Changing data format to support more than 15 devices

laborleben, 2016-02-25:
The header of the FE data words needs to be changed to support more than 15 devices.

Currently:
<1:0> one hot identifier (trigger + TDC) + <1:0> not used + <3:0> FEI4 identifier (without 0, 15 devices) + <23:0> raw data
New Layout:
<1:0> one hot identifier (trigger + TDC) + <5:0> FEI4 identifier (with 0, 32 devices) + <23:0> raw data

Changes are needed in modules pybar_fei4_interpreter, pyBAR Firmware, and pyBAR software.
Keeping compatibility with old raw data files should be possible.

laborleben, 2016-02-25:
This is also necessary for full stave readout.

DavidLP, 2016-02-28:
At the moment it is (if I understand your notation):

<2:0> one hot identifier (trigger + TDC + other device (etc. Mimosa)) + <0:0> not used + <3:0> FEI4 identifier (without 0, 15 devices) + 23:0 raw data

So it could be for 32 FE readout:

<2:0> one hot identifier (trigger + TDC + Mimosa) +  <4:0> FEI4 identifier (without 0, 32 devices) + 23:0 raw data

laborleben, 2016-02-28:
depends on the Mimosa raw data size. Do you know?

themperek, 2016-02-28:
Mimosa raw data is 17 bits ( 1 for start of frame + 16 data). This is waist of bandwidth but this is easiest at the moment.

laborleben, 2016-02-28:
So then the Mimosa data does **not need** an one hot identifier. The one hot identifier for TDC and trigger was used to get user data into the bits below the identifier. E.g. 
trigger: bit 31 is set -> up to 31bits of user data
TDC: bit 30 is set -> up to 30bits of user data (though just 28bits are used, 2bits wasted for the header)

My suggestion now:
For FE and Mimosa data don't make a difference. The mapping can be done in software. We can use the 4 / 5 / 6 bit identifier for both.

The question is, do we need to decide whether we want to connect a maximum of 16, 32 or 64 (sorry for the typo above) devices connected to the MIO3. As long as we don't connect more then 32 devices, the one hot bit 29 is still free to use for future modules (and should not be wasted to the Mimosa readout).

DavidLP, 2016-02-29:
Why don't we need a Other Device header? How to distinguish FE-I4 from Mimosa data then?

themperek, 2016-02-29:
I guess one would need some configuration to know what is FE-I4 and what is Mimosa.
How realistic is to have more then >16 FE-I4?

laborleben, 2016-03-07:
I updated the pyBAR_fei4_interpreter a little. Fixed also a bug there.

Yes, good point, how realistic is >16 chips? Are any specs available for the next stave prototype?

themperek, 2016-03-07:
All "next" will be 65 so everything will change including data format.

laborleben, 2016-03-07:
So 16 chip support is enough.

themperek, 2016-03-07:
I guess if someone needs more then can make a custom version.

laborleben, 2016-04-09:
We came up with the following solution: We want to use the 4bit identifier (channel ID) for both, FEI4 and Mimosa (and any other device that might be added later). This gives us the freedom of any combination and number of up to 16 devices. Device flavor and channel ID have to be set up in the YAML configuration file and stored to the HDF5 raw data file (information needed for the raw data analyzer and online monitor).
In order to simplify the online monitor, the assumption that any channel ID above 9 is assigned to Mimosa telescope planes (up to 6 telescope planes) is maintained.
In addition, Mimosa data can be identified by the empty header (raw data bits [23:16]). Checks for the headers should be added to the software.
The raw data HDF5 file should contain complete data as received from the readout card and/or send right away to the data collector/online monitor without modification/splitting to reduce load of the readout PC (especially important for future applications where the data rate is much higher).

Any comments?

DavidLP, 2016-04-09:
Just another info: This makes it also possible get 2^3 devices with TDC words into the data stream needed for BEAST. Summary (as I understand it):

32 - bit data word
**T**_tdc_ DDDD 0000 0000 0000 0000 0000 0000

**T**: One hot trigger word identifier followed by 31 - bit of data
_tdc_: 3-bit TDC word identifier followed by 28 - bit of data, T = 0
DDDD: 4-bit device identifier followed by 20 - 24 bit of data, all preceding bits = 0

DavidLP, 2016-04-09:
Generally things get complicated when configuration files define the protocol. So I would like to keep the device identifier as constant as possible. Thus to generally assume that 6 bit identify Mimosa and 15 the FE-I4 is fine with me. I do not care about the extreme > 15 FE-I4 use case and all important use cases (BEAST, Mimosa telescope + FE-I4) are possible.

themperek, 2016-04-09:
I would also keep it constant by default if anyone will ever need something else can add configuration.

DavidLP, 2016-08-01:
For  BEAST we did not take into account that 5 TDC modules write into the SRAM that have to have a device identifier bit. Is it possible to reduce the trigger word data from actual 31 bit to 30? Are 31-bit needed for the trigger time stamp?

laborleben, 2016-08-01:
To summarize our discussion:

Beast data should have the following layout:
0x1xxx xxxx xxxx xxxx xxxx xxxx xxxx xxxx  31bit trigger data word
0x0xxx yyyy yyyy yyyy yyyy yyyy yyyy yyyy  TDC data word where x is the 3bit ID starting from 1 and y is 28bit TDC the data
0x0000 yyyy xxxx xxxx xxxx xxxx xxxx xxxx FE data word where y is the 4bit ID starting from 0 and x is the 24bit FE data

This will have no impact on basil and keeps the raw data compatible to the current FEI4 raw data analyzer.

DavidLP, 2016-08-02:
Besides that Mimosa26 and BEAST > 1 #TDC data cannot be distinguished. They would both be seen as OTHER DATA words by the FEI4 raw data analyzer. But that is fine with me.

DavidLP, 2016-11-30:
Since this is fixed now, and I guess even implemented I close this issue

closed 2016-11-30 by DavidLP

laborleben, 2016-11-30:
Nothing was changed, and we have only a hack for the BEAST/FANGS readout and for the Telescope DAQ. So it depends on what you want to achieve in the future.

DavidLP, 2016-12-01:
Which hack? The BEAST readout format was changed.

If I see it correctly we have a format the works with all kind if combinations:
- Mimosa26 + FE-I4
- BEAST Multi FE-I4  + TDC
- FE-I4 only

That is all we need, no?

laborleben, 2017-02-02:
This is done.

commit 6a3ba54c (2017-02-02): ENH: select data from specific channel, prepare for beast and MMC3, fixes #41
commit 3d1795c3 (2017-02-03): ENH: select data from specific channel, prepare for beast and MMC3, fixes #41
commit 87186988 (2017-02-03): ENH: select data from specific channel, prepare for beast and MMC3, fixes #41
commit f792521e (2017-02-03): ENH: select data from specific channel, prepare for beast and MMC3, fixes #41