SiLab-Bonn/pyBAR #48: Stop readout time stamp equals start readout time stamp

DavidLP, 2016-04-10:
It is possible that in the raw meta data start and stop readout time stamp are equal. That does not make sense.

[image]

laborleben, 2016-04-11:
This should not happen. But this might be a problem of the system time.

DavidLP, 2016-04-11:
Is it fixed or understood?

laborleben, 2016-04-11:
No, so I need more information. The fix will just mask the error and it will not show up.

laborleben, 2016-04-11:
To be more clear, it would be better if you revert the fix, otherwise I'm blind to it.

laborleben, 2016-04-11:
Ah you already made a new interpreter version. OK than I will ignore it.

DavidLP, 2016-04-12:
Since we do not rely on the time stamp anywhere, we just leave this bug report open until somebody needs to fix it.