SiLab-Bonn/pyBAR #50: TLU module hangs

DavidLP, 2016-04-11:
Quite often the TLU module issues one trigger and than hangs until a MIO power cycle.

laborleben, 2016-04-11:
More details, which input?

laborleben, 2016-04-11:
It is very unlikely that the complete board hangs when the TLU FSM is stuck. What are the status LEDs telling? Is the USB interface frozen?

DavidLP, 2016-04-12:
### More details

Configuration:
- Trigger input RX2
- Trigger from scintillator
- Trigger inverted
or
- Trigger from TDC (loop through)

Cross checks:
- Trigger / TDC signal is there
- All other scans still work (noise occupancy etc.), that do not require the trigger signal
- Veto does not matter
- There are no trigger words in the data stream
- Nothing (but TLU FSM) is  stuck

Fix:
- Reload FPGA firmware

laborleben, 2016-05-27:
pls test with latest development branch.

laborleben, 2016-06-02:
pls test latest master/development branch with basil v2.4.3. In v2.4.3 there was a minor fix for a bug which could lead to the problem described above.

A possible workaround is to load the FPGA firmware before every run. Change the "avoid_download" parameter in dut_configuration_mio.yaml to False (https://github.com/SiLab-Bonn/pyBAR/blob/master/pybar/dut_configuration_mio.yaml#L5):
```
USB:
    bit_file       : "mio.bit"
    board_id       : 
    avoid_download : False
```

laborleben, 2017-01-06:
I haven't seen that in a while. Any update?

DavidLP, 2017-01-09:
I do not measure so I do not know. Fine to me to close.

laborleben, 2017-01-09:
So most likely the issue was fixed in basil 2.4.3 as stated above.

closed 2017-01-09 by laborleben