SiLab-Bonn/pyBAR #51: Multi chip readout

DavidLP, 2016-04-12:


DavidLP, 2017-02-02:
I think I came across a nice solution how to handle multiple chips with parallel and serial scans. Some thoughts:

- All scans besides the tuning scripts are only writing data to the front-end, the access is done with two handles: [self.register](https://github.com/SiLab-Bonn/pyBAR/blob/multi_chip/pybar/scans/scan_digital.py#L22) and [self.register_utils](https://github.com/SiLab-Bonn/pyBAR/blob/multi_chip/pybar/scans/scan_digital.py#L30)

- These handles are set in the [fei4_run_base](https://github.com/SiLab-Bonn/pyBAR/blob/multi_chip/pybar/fei4_run_base.py)

- The fei4_run_base is the base class of all scans. Thus it can easily know how a scan should be executed (parallel, serial) and it knows about the hardware (one command line broadcast or multiple independent channels)

- We can just define the two handles `self.register` and `self.register_utils` according to the scan/hardware to either 
  1. execute the [command for one chip](https://github.com/SiLab-Bonn/pyBAR/blob/multi_chip/pybar/fei4_run_base.py#L423) ([serial mode](https://github.com/SiLab-Bonn/pyBAR/blob/multi_chip/pybar/fei4_run_base.py#L420) needed for tuning)
  2. execute the [commands for all chips](https://github.com/SiLab-Bonn/pyBAR/blob/multi_chip/pybar/fei4_run_base.py#L441) ([parallel broadcast mode](https://github.com/SiLab-Bonn/pyBAR/blob/multi_chip/pybar/fei4_run_base.py#L439), used for all but tuning scripts for single command line hardware)
  3. execute the commands for all chips at each channel ([parallel channel mode](https://github.com/SiLab-Bonn/pyBAR/blob/multi_chip/pybar/fei4_run_base.py#L308), used for all scripts, but tuning for multiple channel hardware)

With this we do not have to change any scan, besides a flag for the tuning scripts to run serially on modules.

laborleben, 2017-11-15:
This is done.

closed 2017-11-15 by laborleben