SiLab-Bonn/pyBAR #53: using GPAC and FE-I4

owtscharenko, 2016-05-31:
Is there an easy way to read out the FE-I4 with the GPAC ?
In the fei4_run_base usage of the GPAC only with the CCPD seems to be implemented, what was the idea there ?

laborleben, 2016-05-31:
yes, much more simpler than that:

a GPAC firmware is available in the pybar folder (mio_gpac.bit). This is btw the same firmware that is used for the CCPD project.
Just edit the configuration.yaml (https://github.com/SiLab-Bonn/pyBAR/blob/master/pybar/configuration.yaml#L1) and change

```
dut : dut_mio.yaml
dut_configuration : dut_configuration_mio.yaml
```

to

```
dut : dut_mio_gpac.yaml
dut_configuration : dut_configuration_mio_gpac.yaml
```

That's al you need to dol. Please report back if this is working for you.

laborleben, 2016-05-31:
Please note that we updated the master and development branch recently and basil v2.4.3 is required.

DavidLP, 2016-05-31:
If I look into the code I see:

```
# Vdd, also enabling LVDS transceivers
self.dut['CCPD_Vdd'].set_voltage(1.80, unit='V')
self.dut['CCPD_Vdd'].set_enable(True)
if self.dut["CCPD_Vdd"].get_over_current():
  self.power_off()
  raise Exception('Vdd overcurrent detected')
# Vssa
self.dut['CCPD_Vssa'].set_voltage(1.50, unit='V')
self.dut['CCPD_Vssa'].set_enable(True)
if self.dut["CCPD_Vssa"].get_over_current():
  self.power_off()
  raise Exception('Vssa overcurrent detected')
# VGate
self.dut['CCPD_VGate'].set_voltage(2.10, unit='V')
self.dut['CCPD_VGate'].set_enable(True)
if self.dut["CCPD_VGate"].get_over_current():
  self.power_off()
  raise Exception('VGate overcurrent detected')
```

in the case that: ```self.dut.name == 'mio_gpac':```

So is says CCPD, but thats just a variable name?

DavidLP, 2016-05-31:
Ok, the board only provides 2.1 V for regulator powering that is set. The 1.5, 1.8 V for the not existing CCPD will be set, but do not matter. And for direct powering one has to use the Molex connector. I guess that is how is is?

laborleben, 2016-06-09:
I close this because it was a hardware related issue with the FEI4 with through silicon vias.

closed 2016-06-09 by laborleben