SiLab-Bonn/pyBAR #54: Intel MKL FATAL ERROR: Anaconda MKL makes troubles

laborleben, 2016-06-09:
The error was
```
2016-06-08 15:11:11,959 - root - [INFO    ] (MainThread) Start S-curve fit on 2 CPU core(s)
Intel MKL FATAL ERROR: Cannot load libmkl_mc.so or libmkl_def.so.
Intel MKL FATAL ERROR: Cannot load libmkl_mc.so or libmkl_def.so.
Intel MKL FATAL ERROR: Cannot load libmkl_mc.so or libmkl_def.so.
Intel MKL FATAL ERROR: Cannot load libmkl_mc.so or libmkl_def.so.
Intel MKL FATAL ERROR: Cannot load libmkl_mc.so or libmkl_def.so.
Intel MKL FATAL ERROR: Cannot load libmkl_mc.so or libmkl_def.so.
Intel MKL FATAL ERROR: Cannot load libmkl_mc.so or libmkl_def.so.
Intel MKL FATAL ERROR: Cannot load libmkl_mc.so or libmkl_def.so.
^C2016-06-08 15:13:31,313 - root - [INFO    ] (Thread-1  ) Pressed Ctrl-C. Stopping run...
2016-06-08 15:13:31,315 - root - [INFO    ] (Thread-1  ) Pressed Ctrl-C. Stopping run...
```
Even Ctrl-C couldn't stop the run.  The error needs to be invetigated.

A temporary solution is to install conda packages together with "nomkl".

DavidLP, 2016-08-23:
I think this is working at the moment. But who noise what the next release will give us...

DavidLP, 2016-11-28:
I do not have this issues anymore on any PC. So I guess it can be closed.

closed 2016-11-28 by DavidLP