SiLab-Bonn/pyBAR #57: Occupancy plot to pdf

DavidLP, 2016-08-16:
When I use `mask_step: 3` and  `enable_mask_steps: [1]` I get the following result plot:

[image]

How does that make sence?

laborleben, 2016-08-17:
this is just displaying the moiré effect?

DavidLP, 2016-08-17:
Maybe. But I think that was working once.

DavidLP, 2016-08-23:
The mask is correct. It is really just this plot. The larger plot looks also better.

laborleben, 2016-08-24:
https://github.com/matplotlib/matplotlib/issues/2972/

This is explaining it.

DavidLP, 2016-08-24:
I do not see how that explains anything, since I do not see a blur.

laborleben, 2016-08-24:
As it is now in development, the full resolution is displayed. So every pixel is displayed when zooming in. There are effects with some PDF viewers which interpolate when zooming out (the PDF looks like interpolation="nearest").

DavidLP, 2016-08-24:
When I zoom in it looks the same. It is not an interpolation issue when I zoom out ?!

laborleben, 2016-08-24:
The explanation is, that with interpolation="nearest" and using the PDF backend the resolution is reduced which leads to the observed behavior. Yesterday, I changed the "nearest" to "none" and now you are able to see every pixel (when zooming in, at least under Ubuntu with default PDF viewer).

DavidLP, 2016-08-24:
Ahh ok, thanks.

closed 2016-08-24 by laborleben