SiLab-Bonn/pyBAR #58: Tread PlsrDAC ringing

DavidLP, 2016-08-23:
Some FE-I4 (especially after irradiation) show a ringing PlsrDAC:

[image]

 If the injection loop is too quick the injected charge can change leading to measured higher noise and a charge that depends on the injection speed. We should add an additional injection delay with std. value = 0 to all scans using the PlsrDAC to be able to make the injecion speed slower.

laborleben, 2016-08-23:
In the picture there is no ringing before the pulse. Can you explain?

laborleben, 2016-08-23:
OK, I try to explain: The standard delay is around 80us and the next pulse would be still inside the dangerous region (10us/dev). So this plot either show the first pulse or injections with already extended delay.

DavidLP, 2016-08-23:
Long delay. I misscalculated. We can have 400 us injection delay with our current FPGA cmd sequenzer implementation (16000 * 25 ns). That is sufficient. So we just have to add a parameter in software.

DavidLP, 2016-08-23:
Just to make it clear how the analog scan (and thus all other scans) with std. settings (3rd mask) looks like:
[image]

DavidLP, 2016-08-23:
The ringing can be fixed by increasing the OP amp bias of the:

**IV converter op amp**

[image]

responsible for the slow return to baseline.

The configuration for a not ringing PlsrDAC is: `PlsrVgOpAmp = 150`

Disclamer: I think that is this is this OP amps bias. There is also another gain 2 OP amp to increase the rail to rail amplitude. It is not mentioned anywhere, neither in schematics nor the manual. So that is an educated guess.

DavidLP, 2016-08-23:
Since that does not have to be fixed in software I close this issue.

closed 2016-08-23 by DavidLP