SiLab-Bonn/pyBAR #59: PlsrDAC injections too fast so that previous pulse does not return to baseline

laborleben, 2016-08-30:
The left side before the pulse is *NOT* flat.

With default delay (see [here](https://github.com/SiLab-Bonn/pyBAR/blob/31aa34842b1b755e43b9f646426e191bcbda410a/pybar/fei4/register_utils.py#L1267)):

[image]

With increased delay (+1000 clock cycles):
[image]

laborleben, 2016-08-30:
The good thing is that the HitOR Calibration uses a fixed delay of 5000 which already includes a safety margin.

This is also related:
https://github.com/SiLab-Bonn/pyBAR/commit/f0ba0189406511cda2276ef99e4f46c0c6858d9c
https://github.com/SiLab-Bonn/pyBAR/commit/b1e3fed207b7c8583b93c86ca319ac3fc5b2c5f9
https://github.com/SiLab-Bonn/pyBAR/commit/dd8f7cf7dcc13efdeacce86b86efa0ec16ab4d3f

laborleben, 2016-08-30:
Reflection?

[image]

DavidLP, 2016-08-30:
> f0ba018
> b1e3fed
> dd8f7cf

Yes I figured it is needed but did not investigate why :-[

DavidLP, 2016-08-30:
Hehe no. That is crap and no reflection. Does it got when you change PsrDAC bias? Are you sure that you do not avverage over different Pulses. Does a single pulse look like this?

DavidLP, 2016-08-30:
BTW: I never saw this for chips between 60 - 840 MRad. FE-I4 A and B.

laborleben, 2016-08-30:
I get this bump also with cables of different length, grounding and cables types. This is probably a bad timing in the FE. Increasing the delay (repetition speed) in the FPGA removes this bump.

+20 Clocks:
[image]

+60 Clocks:
[image]

+100 Clocks:
[image]

DavidLP, 2016-08-30:
Weird, would be interessting to see the full signal after the injection.

laborleben, 2016-08-30:
I could draw no conclusion other than this is a FE issue. The bumps are moving, getting smaller or larger depending on the delay and PlsrDAC setting but they are never disappearing.

laborleben, 2016-08-30:
PlsrVgOpAmp: 255 -> 100
[image]

laborleben, 2016-08-30:
OK, found the bug: ist was the averaging over all waveforms, which may include other waveforms than the pulse. Fixing the trigger gives me this nice picture (even though I don't know where the 13.5MHz buzz is coming from
[image]
):

laborleben, 2016-08-30:
also very interesting the pulse at PlsrDAC=0:
[image]

This is equivalent to 650e!

themperek, 2016-08-31:
Actually one can try one more time to use external    injection. A proper threshold scan for low thresholds may be possible.

DavidLP, 2016-08-31:
How did you get a measurable signal for PlsrDAC = 0? Can you show the PlsrDAC -Calibration here, espectially for low PlsrDACs that we can see if it is a non linearity or just an offset.

DavidLP, 2016-08-31:
I guess you redued the bandwith, no?

laborleben, 2016-09-01:
This was just changing from edge trigger to pulse trigger and I require a certain pulse minimum length ( O(length pulser signal) ). One of the main problems is, that averaging can distort your results, e.g. averaging a flat line and the signal of interest, would you notice that?
There are probably many ways to fix that. One is to require a certain pulse length and hope that there is no other signal that can rigger the oscilloscope, or just investigate every single pulse and average later (which would increase data taking time).
I also observed that the oscilloscope only triggers on a small fraction of the pulses, e.g. from 512 injections into the FE only ~100 samples were taken by the oscilloscope, even after increasing the delay between the pulses.

What I observed is a high linearity of the pulse. The small non-linearity at the lower end comes most likely from the averaging of the small signal with noise.

Plots will follow, I will address the problems described above and update the script in the coming days.

DavidLP, 2016-09-01:
So summary?: 
 - Transfer function is linear + offset
 - Trigger on pulse is better for small pulse amplitudes

To your averaging question:
- One should reset the averaging storage before using another PlsrDAC setting. This happens when you switch to single shot mode and then back, at least for the oszis we have. This was to my mind also
mentioned in the code.

DavidLP, 2016-11-28:
Since this story has an end I close this issue.

closed 2016-11-28 by DavidLP