SiLab-Bonn/pyBAR #60: GDAC tuning finds 50% occupancy from the noise

laborleben, 2016-08-31:
The GDAC tuning script sometimes finds the 50% occupancy level from noise but not from injection.

laborleben, 2016-09-01:
Sometimes the TdacVbp is too low and needs to be increased to have a lower threshold DAC (GDAC) setting for the target threshold. The algorithm we are currently using is not efficient finding the threshold above GDAC=128.

DavidLP, 2016-09-01:
But the following is true: smaller TdacVbp --> higher threshold -> smaller GDAC after tuning

So how does this fit to: _Sometimes the TdacVbp is too low and needs to be increased to have a lower threshold DAC (GDAC) setting_ ?

laborleben, 2016-09-01:
your statement is right.

DavidLP, 2016-09-01:
Thus decreasing TdacVbp  will make the tuning work again. Can I close this issue?

laborleben, 2016-09-01:
I implemented the background noise check, which works now. To make the noise background check working, the "Enable" mask has to be removed from "enable_shift_masks".

I also want to note that one can change "gdac_tune_bits" from range(7, -1, -1) to range(8, -1, -1) or even higher bits.

closed 2016-09-01 by laborleben

laborleben, 2016-09-01:
Some chips send bad data (8b10b errors) when setting the GDAC to too low values. Therefore I added a lower limit to the GDAC tuning.

This is an example how to set up the tuning to tune most difficult chips:
runmngr.run_run(run=Fei4Tuning, run_conf={"enable_shift_masks": ["C_High", "C_Low"], "gdac_tune_bits": range(7, -1, -1), "max_delta_threshold": 15, "gdac_lower_limit": 40})

DavidLP, 2016-09-01:
That's true. Noisy pixels can disturb communication.