SiLab-Bonn/pyBAR #63: Duplicate event

igjaegle, 2016-09-29:
Events are duplicate. Duplicate seems to always occur in the last event of a readout. It is either a problem at the conversion from word to hit (h5 to interpreted.h5) or at the readout level.

laborleben, 2016-11-23:
The files you send me were containing same events (e.g. event with event_number = 71241 and 75490).
The reason for this is that the raw data already contains the duplicated events. So nothing is wrong with the raw data analyzer.

Can you give more details on your setup? What I can see from the raw data file is that you were using SEABAS2. Do you have the same effect when reading out the FEI4 with USBpix?

igjaegle, 2016-11-23:
Dear Jens,

Thank you for taking the time to look into this, I really appreciate it.

The setup was the following:
Inside the TPC and close from the FE-I4B board there was a LVDS driver (see
enclosed PDF) for the HITOR, since we use the external trigger mode. A
37meter cable between the  SEABAS2 and USBpix
 boards and the TPCs.

The SEABAS2 and USBpix board were inside of two distinct boxes. Inside the
box, the LVDS signal was converted by a receiver (see enclosed PDF).  For
the SEABAS2 before going into the HITOR pin, the signal was passing through
a Xray-veto-circuit.

The duplicate events were only seen with SEABAS2 as far as I can tell.

Thanks, cheers,
 Igal.

themperek, 2016-11-23:
Could you provide full log file?
Nothing interesting there?

PS. For the attachment to propagate you may need tu use web interface: https://github.com/SiLab-Bonn/pyBAR/issues/63

laborleben, 2016-11-23:
OK, thanks for the confirmation. Since I don't have any SEABAS2 in my hand, it is almost impossible for me to do the debugging.
There are huge differences in the design of the readout boards, and I suspect certain interfaces.

igjaegle, 2016-11-24:
Dear Tomasz and Jens,

Thank you for your help. There are log files but I cannot access at the
moment it. Since SEABAS2 will not be re-used with the TPCs, I think there
is no need to solve this problem.

Thanks again, cheers,
 Igal.

closed 2017-01-06 by laborleben