SiLab-Bonn/pyBAR #65: tune_fei4 crashes

DoreenAchnitz, 2016-11-15:
The tuning always crashes due to CmdTimeoutError: Time out while waiting for sending command becoming ready in CMD, module basil.HL.cmd_seq. Power cycle or reset readout board!
- I changed the USB cable and port but it still does not work..

themperek, 2016-11-15:
A full log file could be helpful.

DavidLP, 2016-11-16:
It was actually (as Jens suggested) a FE issue. The threshold setting circuits seems not to work anymore, most likely ESD during handling. A Fifo time out warning was also issued before the _CmdTimeoutError_.
To make me understand: Where is the CMD ready signal connected to. What defines a CMD ready?

laborleben, 2016-11-16:
It might be, that downloading the raw data takes too much time so that reading back the status of CMD gets delayed with the observed results. One might consider increasing the CMD timeout for such cases.

DavidLP, 2016-11-16:
So the signal is set in the command sequencer but the reading back via usb is delayed. Thus it is a threading issue?

laborleben, 2016-11-16:
Actually  I don't know. I don't fully understand the chain of failures yet. Need to have a closer look.

DavidLP, 2016-11-16:
@DoreenAchnitz: Can you post the complete error/warning log?

laborleben, 2016-11-16:
I fixed a bug in the tune_fei4.py meta script. The lower threshold (GDAC) limit can be set up, so that we can avoid going too low with the GDAC and prevent the chip becoming noisy.

DavidLP, 2016-11-28:
Still crashing. Also shows warning:

```
2016-11-28 13:20:02,882 - root - [INFO    ] (MainThread) Resetting SRAM FIFO: size = 0
2016-11-28 13:20:03,082 - root - [WARNING ] (MainThread) SRAM FIFO not empty after reset: size = 4
```

How is that possible?

DavidLP, 2016-11-28:
So I found the bug introduced in d8b6dff.
It is the bad calculation of software delays that make this PC crash. 

If I look for example at:
[this](https://github.com/SiLab-Bonn/pyBAR/blame/development/pybar/fei4/register_utils.py#L113): ` delay = length * 25e-9 * repeat - 0.002  # subtract 2ms delay`
I feel sick.

2 ms fixed subtraction for a value that is already potentially small is very bad programming. Why?!

closed 2016-11-28 by DavidLP

laborleben, 2016-11-28:
You don't need to become sick, it's just a bug. This was because of the USB read/write delay which is roughly 1-2ms. This will increase the CMD speed, because the check for CMD ready starts earlier. Reducing the delay by 0.2 ms is not the error, the error was couple of lines below:
```
            if delay is None:
                timeout = 1
```
has to become
```
            if delay is None or delay < 0.0:
                timeout = 1
```

Be aware that your commit 2c14e8bd5831b4a63200bdda9a10a3a5debf4282 will make CMD slower.

laborleben, 2016-11-28:
Concerning 576e4c1bc6a2b35c2625f4a962e05c6d5a14614e:

A better solution would be having a minimal timeout value. 100 times a small value is still small. I suggest to have a minimum timeout of 1 second (timeout=1.0) and adding the calculated command length to the minimum timeout.

DavidLP, 2016-11-28:
`Reducing the delay by 2ms is not the error`: Not `really, the delay was positive, I checked.

DavidLP, 2016-11-28:
Potentially slightly slower but working is fine with me.

DavidLP, 2016-11-28:
Your check shoud be:
``` 
if delay is None or delay <= 0.0:
    timeout = 1
```

laborleben, 2016-11-28:
but what was the error message exactly? I don't get the full picture.

laborleben, 2016-11-28:
The delay can be 0 but should not crash the readout.

DavidLP, 2016-11-28:
What happens is that we run into a time out because of a too small calculated delay value. 

1. The exception is raised in basil [here](https://github.com/SiLab-Bonn/basil/blob/master/basil/HL/HardwareLayer.py#L59)

2. Then the exception is catched, rebranded and rethrown [here](https://github.com/SiLab-Bonn/pyBAR/blob/master/pybar/fei4/register_utils.py#L134), wich makes it impossible to see where it actually went wrong. 

3. So your error message is `Time out while waiting for sending command becoming ready ....`

laborleben, 2016-11-28:
1. I have submitted a fix for negative timeout values. A negative timeout will throw an exception: https://github.com/SiLab-Bonn/basil/commit/22740a48e5d0422db9d77274c3fad0276286b0e0
2. Might not be a good idea, but in the first place I thought that this new exception gives a better explanation for the occurring error. If we keep the original exception it will just point to the generic basil module.

DavidLP, 2016-11-28:
For properly solving 2. (original exeption and additional user information) we need a [Python 3 feature](https://blog.ionelmc.ro/2014/08/03/the-most-underrated-feature-in-python-3/).
In python 2 I would never rebrand exeptions. One could catch them, add error info, and then rethrow.

laborleben, 2016-11-28:
Good link. As far as I know you can keep the traceback and change the message in python 2. These kind of dirty tricks I play in https://github.com/SiLab-Bonn/pyBAR/blob/master/pybar/fei4_run_base.py#L375 and following.

laborleben, 2017-01-06:
Doreen, can you check?

DoreenAchnitz, 2017-01-09:
So far no problems occured!

laborleben, 2017-01-09:
OK, thanks.