SiLab-Bonn/pyBAR #66: ViTables v2.1 incompatibel with pytables 3.3

laborleben, 2016-11-22:


DavidLP, 2016-11-28:
But that is no solution for a setup with Qt5, no?

laborleben, 2016-11-28:
Not yet tested with Qt5. The solution above is also not satisfying. ViTables just needs more support.

DavidLP, 2016-11-28:
ViTables is another dead python package. Changing to Qt5 would take a couple of hours.
Internetfame, anybody?

DavidLP, 2016-11-30:
A work around is to setup a python environment with QT4 for vitables only.

```
conda create -n vitables python=2.7 pyqt=4.11.4 pytables
source activate conda vitables
pip install https://github.com/uvemas/ViTables/archive/v2.1.zip
```

laborleben, 2017-01-06:
Seems like this is the only viable solution right now.

laborleben, 2017-06-09:
This was recently fixed in ViTables `master` branch.

Use `python setup.py install` to install vitables.

closed 2017-06-09 by laborleben