SiLab-Bonn/pyBAR #74: Git screw up

DavidLP, 2017-02-03:
To revert a mistake in the `multi_chip` branch I force pushed via `git push --force` which led to:

`Total 0 (delta 0), reused 0 (delta 0)
To https://github.com/SiLab-Bonn/pyBAR
 + 5fafcd3...4dd3587 development -> development (forced update)
 + d8ed5cd...a7edc82 multi_chip -> multi_chip (forced update)`

Unfortunately that does not only force a reset on the actual branch but on all branches (wtf?). So our development has now a lot of missing commits.

DavidLP, 2017-02-03:
By the way due to the extensive use of force update and commit picking and whatever github black magic our time line is not linear anymore ...

DavidLP, 2017-02-03:
A fixed development branch with correct last `development` commit hash `5fafcd3` is created with the name `dev`.

laborleben, 2017-02-03:
I fixed development. I don't know what happened. But I had on my disk the full timeline and I did push force. The full timeline is recovered now.

DavidLP, 2017-02-03:
Merci

closed 2017-02-03 by DavidLP

DavidLP, 2017-02-05:
Still something seems wrong, because travis thinks now that `mult_chip` is `development`. For whatever reason.

laborleben, 2017-02-05:
We'll remove that branch after merging. I can't see any problems in the timeline using git command line. All branches are fine.