SiLab-Bonn/pyBAR #75: USBError: No such device

derekstrom, 2017-03-08:
Reporting the following USB error, which we observe during tunefei4 scan.

2017-02-27 15:22:50,092 - root - [WARNING ] (ReadoutThread) Recovering from the USB error, repeating ReadExternal in SiUsb3
2017-02-27 15:22:51,101 - root - [ERROR   ] (ReadoutThread) [Errno 19] No such device (it may have been disconnected)
2017-02-27 15:22:51,101 - root - [INFO    ] (MainThread) Stopped FIFO readout

Software
Pybar release: multi
Basil release: mio3

PC
Lenovo ThinkCentre M93 Desktop PC
• Intel Core i3-4170 CPU, 3.70 GHz
• 20 GB RAM
• USB 3.0 ports
• OS: Windows 7 Professional 64-bit

Hardare
- USBPix3 (Board #020)
- USB 3.0 cable connected to PC
- 2x Single-chip FE-I4 SCC v1.1
- Cat 6 Ethernet cable

laborleben, 2017-03-09:
Hi, we are currently working on a new firmware for the MMC board that uses Ethernet to connect to the host PC.

I'll leave this ticket open until we finished the task.

laborleben, 2017-03-09:
For the time being the firmware for the MMC that support the 6x Mimosa26 and a single FEI4 plane can be used (if the readout of a single FEI4 is enough). If needed, we can provide instructions on how to set up pyBAR to use Mimosa26+FEI4-firmware (Mimosa26 not necessarily connected).

laborleben, 2017-03-09:
@FlorianHinterkeuser: can you comment how far you are with the implementation?

FlorianHinterkeuser, 2017-03-09:
Hi, 
in short: as of now readout of up to five single FE-I4 (assuming single modules & one FE per channel) can be done with a slightly derived version of the MMC3 beast firmware (can be found in the multi_chip_sp branch).

themperek, 2017-03-09:
@derekstrom A initial version that suport 8 FE you can find here: https://github.com/SiLab-Bonn/pyBAR/tree/multi_chip_sp/firmware/mmc3_quad_eth We will be working on this.

Could you describe your needs?

derekstrom, 2017-03-10:
Hi,

Our current needs are for support up to 4 FE. The multi_chip_sp seems to fit that. Thanks.

themperek, 2017-03-18:
@derekstrom Did you manage to run it?

laborleben, 2017-04-06:
@derekstrom : Please be aware that we are currently working on the multi_chip_sp branch of pyBAR. Please keep the branch up-to-date.

closed 2017-04-06 by laborleben