SiLab-Bonn/pyBAR #79: Independent triggering

trogonpete, 2017-04-06:
Is it possible to accept multiple _independent_ external triggers (HitOR) from multiple FE-I4 chips on a single MMC3 board? If so, how many chips are independently triggerable?

trogonpete, 2017-04-06:
Do the FE-I4s on the FANGS staves not trigger independently?

We will have 8 FE-I4s reading out uncorrelated and asynchronous events, so all we can do with parallel triggering is trigger all 8 chips from the OR of all 8 HitORs. At high data rates, this seems like it will be quite inefficient. Can you foresee any solutions that might make independent triggering possible? TLU magic?

What about with FE-I4 self-trigger scans? Is asynchronous triggering/readout possible? (but we ideally need to use stop-mode scans for more than 16 BCs)

themperek, 2017-04-07:
I can imagine a hack that may work but this is bit of work. To have 8 cmd_seq that have same address in the fpga and receive same data they can then be triggered independently. It would nee to be connected in way that only one modules is read back, If every chip has different address even even tuning/configuration may work ...

DavidLP, 2017-04-07:
> What about with FE-I4 self-trigger scans? Is asynchronous triggering/readout possible?

Sure that is possible, but not with more than 16 BCIDs.

trogonpete, 2017-04-08:
OK, it's good to know at least that this is possible in firmware. Hardware-wise, how would HitOR be received by MMC3? I suppose we'd need to build a board that receives the 8 HitOR lines and talks to the TLU? (I guess this is probably done on some of the existing FE-I4 boards but we're using a homebrew one)

laborleben, 2017-04-08:
The firmware/software can modified to support multiple FE individually. But that needs significant changes in general. If you specify your use case more precisely a bespoke solution exist that doesn't need severe changes of the current system. If @themperek is right, and only slight changes of the firmware/software might be necessary. We have to look into that.

trogonpete, 2017-04-09:
Our use case (for BEAST phase II):

1. 8 independent time projection chambers (TPC)
2. One FE-I4B inside each TPC on a custom board (nothing fancy: standard LVDS via RJ-45, HitOR via LEMO, power via Molex, chip ID is settable via headers)
3. For each TPC, we have 2x15m cat7 cables to transmit power, receive HitOR (converted to 3.3V logic) and communicate via LVDS
4. One cat7 cable from one TPC is plugged into one of the 2x4 RJ-45 jacks on MMC3
5. HitOR (converted to 3.3V) is plugged into RX0 LEMO jack (obviously we can only do this for one chip at the moment)
6. DAQ via stop mode scans (right now we're using mmc3_r26_eth firmware, but modified to accept triggers on RX0 instead of TLU). Our tracks are in general much longer than 16 BCIDs, so FE-I4 self-trigger scans are not appropriate. 
7. Communication with computer (SL6.4) via SiTCP

Important operational points:

a. There is no time correlation between tracks in different TPCs. That is, using a common trigger means that 7 of the 8 TPCs will be reading out nothing. 
b. The firmware solution to rejecting short HitOR triggers is essential for our use case. Otherwise we're absolutely swamped with backgrounds. 
c. We can't change anything at this point inside the TPCs. So if we need to add some hardware to talk to the TLU, this will have to be done next to the MMC3 board.
d. The trigger rate for one TPC is on the order of 10 Hz maximum. 

I hope this paints a clearer picture. Thanks!

trogonpete, 2017-04-27:
So, does this sound doable?

themperek, 2017-04-27:
It should be possible.  Would be easier to have HitOr in the RJ45 cable (like BEAST).

trogonpete, 2017-04-28:
No problem. Can the RJ45 pinout handle 8 single-ended HitOR inputs?

laborleben, 2017-04-28:
Yes, all HitOr via 1 single Ethernet cable (8x single ended) is possible or 8 differential pairs (similar to the BEAST/FANGS, which I prefer). @PatrickAhl We probably need a similar solution for BEAST/FANGS anyway.

laborleben, 2017-04-28:
@trogonpete I prefer a solution where we have clock, CMD, data and HitOR bundled in a single Ethernet cable for each FEI4 readout chip. An individual CMD line is necessary to independently trigger the FEI4 chips. The clock can be shared among the FEI4. But we never connected that many chips to a single output buffer. So I prefer also individual clock for each FEI4 chip.
We need to check if we can fit the I/O (8x 4x differential pairs) into the FPGA.

DavidLP, 2017-05-12:
### Summary from meeting on 12.05.2017

- Requirements:
  - MMC3 readout for 8 ATLAS-FEI4
  - Each triggered independently via hitbus
  - Stop mode readout with 10 Hz trigger rate
  - Hitbus trigger veto for pulses between 50 - 500 ns (2 - 20 clock cycles with 40 MHz)
  - Compatibility to differential 3.3 V Hitbus signal
  - One patch cable per device with CMD, CLK, DATA, HITBUS
- BEAST FANGS have different synchronous trigger
- PyBAR multichip branch to be extended by additional BEAST TPC firmware
- Programming of XILINX FPGA needed to be able to test different firmwares
- Timeline
   - First single chip readout with correct pin assignment (few days)
   - Then async triggering with 8 x (CMD + TRG) module in FPGA (end of July)

laborleben, 2017-07-26:
@trogonpete I've some updates for you. We updated the multi_chip branch and have the firmware for 8x single chips ready and tested. The latest reference PCB design for the interconnectivity (CMD, CLK, DATA, HITBUS/HitOR) is available here: http://icwiki.physik.uni-bonn.de/svn/USBpixI4/pcb/SCC_FE-I4B/SCC_FE-I4B_v2-11/

Missing items are:
- async triggering with 8 x (CMD + TRG) module in FPGA
- so far we can only trigger on the first 7 modules. The 8th module has HITBUS/HitOR not connected and cannot be used for trigger

Both issues are due to limited space in the data header. These features require some more thinking how to do the implementation. The implementation should have minimal impact on firmware and software.

trogonpete, 2017-07-31:
Excellent! 

Unfortunately, I can't open the link ("authorization required") so I'm not sure how to wire the RJ-45 plug. We will be ready to test immediately once I can figure this out.

laborleben, 2017-08-02:
@trogonpete I've found a solution for the DAQ code and have code almost ready for independent triggering. I'll will soon finish a firmware that is supporting independent triggering of several modules.

laborleben, 2017-11-06:
This is done and available in the multi_chip branch as of now!

closed 2017-11-06 by laborleben