SiLab-Bonn/pyBAR #83: Progressbar2 change

DavidLP, 2017-09-28:
With the change to `progressbar2` in the `multi_chip` branch errors will occur in all our other projects, including `pybar` `development`.
The issue is that the old package `progressbar_latest` is used in all other projects. When having  `progressbar_latest` and `progressbar2`, both modules have to be imported via `import progressbar`, so cannot be used at the same time. Additionally their `API` is not compatible (e.g.: `max_value` variable). I think we should refrain from `progressbar2` usage, since it will create unnecessary work and at the end it is just a progress bar. Not really important.
If the urge is there to use another package one should use the _much_ better `tqmd`. That avoids the name clash.

laborleben, 2017-09-28:
Yes, I'll fix that with a rebase. I'll do that later when pushing other changes.

closed 2017-09-28 by laborleben

laborleben, 2017-10-06:
Removed progressbar2 from multi_chip branch.