SiLab-Bonn/pyBAR #84: Threaded scans lead to crashing fei4_interpreter

FlorianHinterkeuser, 2017-10-17:
Running multithreaded scans using the multi-tx FW leads to crashes during data interpretation:

Traceback (most recent call last):
  File "C:\Users\Florian\git\pyBAR\pybar\run_manager.py", line 156, in run
    self.do_run()
  File "C:\Users\Florian\Miniconda2\lib\contextlib.py", line 24, in __exit__
    self.gen.next()
  File "C:\Users\Florian\git\pyBAR\pybar\run_manager.py", line 219, in _run
    self.post_run()
  File "C:\Users\Florian\git\pyBAR\pybar\fei4_run_base.py", line 810, in post_run
    self.analyze()
  File "C:\Users\Florian\git\pyBAR\pybar\scans\scan_digital.py", line 43, in analyze
    analyze_raw_data.interpret_word_table()
  File "C:\Users\Florian\git\pyBAR\pybar\analysis\analyze_raw_data.py", line 632, in interpret_word_table
    self.interpreter.set_meta_data(self.meta_data)  # tell interpreter the word index per readout to be able to calculate the event number per read out
  File "pybar_fei4_interpreter\data_interpreter.pyx", line 132, in pybar_fei4_interpreter.data_interpreter.PyDataInterpreter.set_meta_data
    self.thisptr.setMetaDataV2(<MetaInfoV2*&> meta_data.data, <const unsigned int&> meta_data.shape[0])
IndexError: Time stamp not increasing.

DavidLP, 2017-10-27:
Timestamps that are not increasing cannot be. This has to be fixed in _data writing_ to disc. The subsequent exception when interpreting is wanted and good.

laborleben, 2017-10-27:
This can be closed. This was a bug in the multi_chip branch of pyBAR and is fixed now.

closed 2017-10-27 by laborleben