SiLab-Bonn/pyBAR #85: Installation issue 'UnicodeDecodeError: 'ascii' codec can't decode byte' for pyBAR_fei4_interpreter@development

OlivierHartmann, 2017-11-17:
Hi,
I have a problem when installing pyBAR_fei4_interpreter@development  with command:
`sudo -H pip install git+https://github.com/SiLab-Bonn/pyBAR_fei4_interpreter@development`

I get this traceback:
```
Collecting git+https://github.com/SiLab-Bonn/pyBAR_fei4_interpreter@development
  Cloning https://github.com/SiLab-Bonn/pyBAR_fei4_interpreter (to development) to /tmp/pip-jtkmvS-build
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-jtkmvS-build/setup.py", line 70, in <module>
        ext_modules=cythonize(extensions),
      File "/usr/local/lib/python2.7/dist-packages/Cython/Build/Dependencies.py", line 920, in cythonize
        aliases=aliases)
      File "/usr/local/lib/python2.7/dist-packages/Cython/Build/Dependencies.py", line 815, in create_extension_list
        kwds = deps.distutils_info(file, aliases, base).values
      File "/usr/local/lib/python2.7/dist-packages/Cython/Build/Dependencies.py", line 670, in distutils_info
        return (self.transitive_merge(filename, self.distutils_info0, DistutilsInfo.merge)
      File "/usr/local/lib/python2.7/dist-packages/Cython/Build/Dependencies.py", line 680, in transitive_merge
        node, extract, merge, seen, {}, self.cimported_files)[0]
      File "/usr/local/lib/python2.7/dist-packages/Cython/Build/Dependencies.py", line 691, in transitive_merge_helper
        for next in outgoing(node):
      File "/usr/local/lib/python2.7/dist-packages/Cython/Utils.py", line 44, in wrapper
        res = cache[args] = f(self, *args)
      File "/usr/local/lib/python2.7/dist-packages/Cython/Build/Dependencies.py", line 607, in cimported_files
        pxd_file = self.find_pxd(module, filename)
      File "/usr/local/lib/python2.7/dist-packages/Cython/Utils.py", line 44, in wrapper
        res = cache[args] = f(self, *args)
      File "/usr/local/lib/python2.7/dist-packages/Cython/Build/Dependencies.py", line 591, in find_pxd
        pxd = self.context.find_pxd_file(relative, None)
      File "/usr/local/lib/python2.7/dist-packages/Cython/Compiler/Main.py", line 239, in find_pxd_file
        pxd = self.search_include_directories(qualified_name, ".pxd", pos, sys_path=sys_path)
      File "/usr/local/lib/python2.7/dist-packages/Cython/Compiler/Main.py", line 280, in search_include_directories
        tuple(self.include_directories), qualified_name, suffix, pos, include, sys_path)
      File "/usr/local/lib/python2.7/dist-packages/Cython/Utils.py", line 30, in wrapper
        res = cache[args] = f(*args)
      File "/usr/local/lib/python2.7/dist-packages/Cython/Utils.py", line 148, in search_include_directories
        path = os.path.join(dir, dotted_filename)
      File "/usr/lib/python2.7/posixpath.py", line 73, in join
        path += '/' + b
    UnicodeDecodeError: 'ascii' codec can't decode byte 0xc3 in position 17: ordinal not in range(128)
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-jtkmvS-build/
```
I have installed all required packages before.
What's wrong ?  I'm on Ubuntu 16.04 LTS.
Thank you

laborleben, 2017-11-17:
You are using your system's python which is located in ```/usr/local/lib/python2.7```. The Python that comes with Ubuntu may come with (outdated) packages (Cythonm, numba, etc.) that are not anymore compatible with our software.
We recommend using Anaconda/Miniconda Python which can be installed locally into your home folder  and doesn't interfere with your system's Python interpreter.

laborleben, 2017-11-17:
Can you post ```pip list```?

OlivierHartmann, 2017-11-17:
Hi, thanks for replying.

```
 pip list
alabaster (0.7.10)
asn1crypto (0.22.0)
Babel (2.5.0)
backports-abc (0.5)
backports.functools-lru-cache (1.4)
backports.shutil-get-terminal-size (1.0.0)
backports.ssl-match-hostname (3.5.0.1)
bitarray (0.8.1)
certifi (2017.7.27.1)
cffi (1.10.0)
chardet (3.0.4)
conda (4.3.30)
cryptography (2.0.3)
cycler (0.10.0)
Cython (0.26.1)
decorator (4.1.2)
docutils (0.14)
enum34 (1.1.6)
funcsigs (1.0.2)
functools32 (3.2.3.post2)
idna (2.6)
imagesize (0.7.1)
ipaddress (1.0.18)
ipython (5.4.1)
ipython-genutils (0.2.0)
Jinja2 (2.9.6)
llvmlite (0.20.0)
MarkupSafe (1.0)
matplotlib (2.1.0)
mock (2.0.0)
nose (1.3.7)
numba (0.35.0+10.g143f70e90.dirty)
numexpr (2.6.2)
numpy (1.13.3)
pathlib2 (2.3.0)
pbr (3.1.1)
pexpect (4.2.1)
pickleshare (0.7.4)
pip (9.0.1)
prompt-toolkit (1.0.15)
ptyprocess (0.5.2)
pycosat (0.6.2)
pycparser (2.18)
Pygments (2.2.0)
pyOpenSSL (17.2.0)
pyparsing (2.2.0)
pyserial (3.4)
PySocks (1.6.7)
python-dateutil (2.6.1)
pytz (2017.2)
PyYAML (3.12)
pyzmq (16.0.2)
requests (2.18.4)
ruamel-yaml (0.11.14)
scandir (1.6)
scipy (1.0.0)
setuptools (36.5.0.post20170921)
simplegeneric (0.8.1)
singledispatch (3.4.0.3)
six (1.10.0)
snowballstemmer (1.2.1)
Sphinx (1.6.3)
sphinxcontrib-websupport (1.0.1)
subprocess32 (3.2.7)
tables (3.4.2)
tornado (4.5.2)
traitlets (4.3.2)
typing (3.6.2)
urllib3 (1.22)
wcwidth (0.1.7)
wheel (0.29.0)
```

OlivierHartmann, 2017-11-17:
Sorry, I don't really understand. If I should use miniconda only, how should I install pyBAR_fei4_interpreter with it then?

laborleben, 2017-11-17:
I set up the pyBAR_fei4_interpreter with the command form above and it worked like a charm. I used Ubuntu's (16.04) Python interpreter and the packages on my system had the same version numbers as in your ```pip list``` printout.

Can you check if there is no other Cython than Cython==0.26.1 in your ```/usr/local/lib/python2.7/dist-packages/``` folder? Maybe remove (```rm -r```) any occurrence of Cython and reinstall Cython via pip.

If you are using Anaconda/Miniconda, the commands are exactly the same (but without ```sudo```). To use Anaconda/Miniconda, the Anaconda/Miniconda path has to be added to the PATH variable (this can be done during installation of Anaconda/Miniconda). Please check the README.md of the corresponding branch. Just copy and paste the commands.

OlivierHartmann, 2017-11-21:
Hi,
I have done what you said but it changed nothing.
But it's not grave, I found another way to do what I wanted.
Thanks anyway.

closed 2017-11-21 by laborleben