SiLab-Bonn/pyBAR #86: Binary search algorithms could be slightly improved

DavidLP, 2017-11-19:
There is one special case where we could find a better value:
Imagine:
- We are at the last iteration: `TDAC = 1111` with `Occ = N_inj`: means threshold is still too low
- Then we do the additional last bit = 0 iteration: `TDAC = 1110` with `Occ = N_inj - 1`. Here we measure just by statistical error one count less. Than this result is taken. But we know already that even `TDAC = 1111` was still too low bad. This information could be used to prevent this.

Solution:
Do not do / use the last iteration result when all preceding bits are set.

laborleben, 2018-10-31:
Don't we have these issues with every scan step? The issue is more likely to appear for the LSB, but also the tuning error is the smallest for the LSB being wrong.
I don't see how we can solve this and get significant better results.

A solution would be to increase the number of injections - this is equivalent to increased statistics.

closed 2018-10-31 by laborleben

DavidLP, 2018-11-01:
Significant better results not, but a cross check that threashol(TDAC) is monotone could inprove the result.

commit 0bb13b80 (2018-10-31): ENH: optimize last step in pixel tuning, fixes #86
commit b653f979 (2018-10-31): ENH: optimize last step in pixel tuning, fixes #86