SiLab-Bonn/pyBAR #89: ipython crash upon import

owtscharenko, 2018-05-03:
using `from pybar import ` with autocomplete results in

> 2018-05-03 12:58:16.249 python[32408:3974038] !!! BUG: The current event queue and the main event queue are not the same. Events will not be handled correctly. This is probably because _TSGetMainThread was called for the first time off the main thread.

I narrowed it down to following line in pybar.__init__.py 👍 
`from pybar.scans import *`

owtscharenko, 2018-05-03:
we found the actual issue, `matplotlib.pyplot ` was imported in 3 scans:

- `calibrate_plsr_dac_transient`
- `tune_tlu`
- `scan_iv`

also importing polyfit and poly1d from pylab (in `calibrate_plsr_dac_transient` )cause this issue.

laborleben, 2018-05-03:
Thanks, this is fixed in c3a14f1ab32be488e8a3b198af4678b4664dbbe8, f10ebb68e5aa1000c28fc6de190ad00d070239ce, and 0d024d22391e38e2cb9e75e6df873db1ac73b259.

closed 2018-05-03 by laborleben