SiLab-Bonn/pyBAR #90: Remove ViTables from required packages in Step-by-step installation guide

leloup314, 2018-05-09:
[ViTables](https://github.com/uvemas/ViTables/archive/v2.1.zip) is deprecated and depends on `PyQt4` which is no longer up-to-date and will not work. Rather install  Vitables system wide via `sudo apt-get install vitables`

laborleben, 2018-05-09:
```sudo apt install vitables``` will install an outdated vitables version, so this is not recommended.

Rather use vitables from github:
```git clone https://github.com/uvemas/ViTables.git```
If you use Anaconda/Miniconda, edit ```setup.py``` and remove the PyQt5 dependency (remove the line with ```'PyQt5 (>=5.5.1)',```) and run
```conda install qtpy pyqt pytables numpy numexpr```.
Finally install with
```python setup.py install```.
Done!

leloup314, 2018-05-09:
Then this should be the way to go. Either way the Step-by-step installation guide should be updated.

laborleben, 2018-05-09:
Done!

closed 2018-05-09 by laborleben