SiLab-Bonn/pyBAR #91: run started in thread returns run_status none

owtscharenko, 2018-07-09:
After starting a run with `use_thread=True` and calling the status with a timeout eg. `join(0.01)` it returns `none`. Reading the code I think the status should be RUNNING. When the run is finished, the correct status is returned (CRASHED / FINISHED / ABORTED / STOPPED).

laborleben, 2018-07-09:
```join()``` returns the status of the run after the run has finished. ```None``` is returned because no status is available before the scan has finished. This can lead to ambiguous situations where the return value is ```None```. 

Adding a parameter to ```thunkfy()``` to set up a default return value, would improve the situation.

owtscharenko, 2018-07-09:
I see. But now RunManager crashes in threaded mode:

>   File "pybar/run_manager.py", line 562, in run_run_in_thread
>     return run.run(run_conf=local_run_conf, default=self.current_run.run_status)
> TypeError: run() got an unexpected keyword argument 'default'

this is fixed by adding `default = None`  in `run` method of `RunBase` (L. 137)

Nevertheless, None is still returned when calling a running run

laborleben, 2018-07-09:
OK, this is fixed. I rebased the development branch.

commit acb1a77a (2018-07-09): ENH: adding default return parameter to join(), now returns run status, fixes #91
commit 68cf3848 (2018-07-09): ENH: adding default return parameter to join(), now returns run status, fixes #91
commit bec19da0 (2018-07-10): ENH: adding default return parameter to join(), now returns run status, fixes #91
closed 2018-07-10 by laborleben