SiLab-Bonn/pyBAR #93: Incorrect exception when having multi RX/TX?

DavidLP, 2019-09-07:
I have two modules with independent RX/TX connected via MMC3. I get an exception telling me that I have to set a chip address.
```
ValueError: Parameter 'chip_address' must be different for each module in module group TX=CMD_CH0_TO_CH7.
```
This should not be needed, no?

laborleben, 2019-09-07:
This is/was there for preventing setting up a wrong configuration. This might not be necessary for a preliminary test. But for some scans (e.g., `scan_ext_trigger.py`) you will end up with the wrong configuration for all but one chip. For general usage, for which we use this branch, **I do not recommend** removing the exception because we can damage chips by providing a wrong configuration. 

Can you check the follwoing:

In `fei4_run_base.py` comment the last 2 lines in this code block:
```Python
chip_addresses = list(set([module_cfg['chip_address'] for module_id, module_cfg in self._module_cfgs.items() if module_id in module_group]))
# if len(module_group) != len(chip_addresses) or (len(module_group) != 1 and None in chip_addresses):
#     raise ValueError("Parameter 'chip_address' must be different for each module in module group TX=%s." % tx)
```

This should allow you to use any combinations of chip addresses and even broadcast (chip_address=None)

laborleben, 2019-09-10:
I checked what can be done about this issue... there is no easy/quick solution. Any changes improving that situation would require a significant code change.
**Your issue can be easily solved by using the multi-TX firmware.** The multi-TX firmware is available for the MMC readout board and requires little changes of your configuration file.

closed 2019-09-10 by laborleben